OpenCPN/OpenCPN #3756: WMM: Wrong variation

leamas, 2024-03-26:
**Describe the bug**
WMM shows wrong variation 11° instead of expected 9° in my home area.

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpenCPN, enable WMM plugin
2. Go to position 65° 35,0 N 22° 11,0 E
3. See WMM displaying 11° variation

**Expected behaviour**
9°

**Additional context**
This is close to the pole and thus quite challenging. If we cannot provide a correct value due to problems with the algorithms it's probably better to fail visibly, not providing anything at all.

This is a drive-by comment, I'm not actually using this.

nohal, 2024-03-26:
How should we know what is on the paper chart, that it is right and that it is better than the result of the WMM model to know that we should not show anything at these coordinates?

leamas, 2024-03-26:
Good question. I don't have a quick answer.

nohal, 2024-03-26:
There is no way to do it AFAIC. That the model is 15% off in the last year of it's validity is probably no issue, affects absolutely everything using the WMM model and not considered a problem by anybody. If the geomagnetic science community does consider the results to be too off they release an unscheduled update as they did with WMM2015. Nothing like that is happening for WMM2020 and WMM 2025 will happen on the normal schedule this December.

leamas, 2024-03-26:
OK, I get it. Closing, basically a WONTFIX.

closed 2024-03-26 by leamas