OpenCPN/OpenCPN #3757: SignalK Server Discovery is broken

TwoCanPlugIn, 2024-03-26:
**Describe the bug**
SignalK Server Discovery does not work in version 5.8.4. It did work in version 5.6.x.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options->Connect->Add Connection
2. Select Network
3. Select SignalK
4. Check "Automatic Server Discovery"
5.  Press Discover Now

**Expected behavior**
If a mDNS Server is present, and configured with a valid service record, OpenCPN should issue a mDNS (aka Bonjour) query over UDP Port 5353 and configure a valid SignalK Server connection.

OpenCPN v5.6 worked as expected. OpenCPN  v5.8.4 does not transmit a mDNS query and this does not automatically configure a valid SignalK Server connection

**Screenshots**
OpenCPN v5.6, working as expected.

[image]
[image]

Network Monitor Capture displaying corresponding UDP port 5353 traffic
[image]

OpenCPN 5.8.4 Does not discover SignalK Server entries. No UDP Port 5353 traffic is observed.

[image]

This is the output from teh avahi browse command, displaying the SignalK Web Socket Service Record.
`avahi-browse _signalk-ws._tcp -t -r`
`+   wlo1 IPv6 SignalK on stevead-ubuntu                     _signalk-ws._tcp     local`
`+   wlo1 IPv4 SignalK on stevead-ubuntu                     _signalk-ws._tcp     local`
`+     lo IPv4 SignalK on stevead-ubuntu                     _signalk-ws._tcp     local`
`=   wlo1 IPv6 SignalK on stevead-ubuntu                     _signalk-ws._tcp     local`
 `  hostname = [DESKTOP-IKVUVTU.local]`
 `  address = [192.168.1.104]`
 `  port = [3000]`
 `  txt = ["roles=master,main" "txtvers=1"]`
`=   wlo1 IPv4 SignalK on stevead-ubuntu                     _signalk-ws._tcp     local`
`   hostname = [DESKTOP-IKVUVTU.local]`
`   address = [192.168.1.104]`
`   port = [3000]`
 `  txt = ["roles=master,main" "txtvers=1"]`

**Desktop (please complete the following information if applicable):**
 Only tested on Windows, with the avahi server running on an Ubuntu Linux PC on the same subnet.

closed 2024-04-08 by nohal

TwoCanPlugIn, 2024-04-09:
Yes, it now finds the local SignalK Server, however I think closing this now might be a little premature.

From a UI perspective the "Discover Local Server..." button, shouldn't have the three dot suffix.
Three dots on a button or menu item means that a dialog is shown. (for example a Browse button that shows the File Open dialog). 
A button or menu item without the three dots, just means that an action is performed (for example OK and Cancel buttons). 

From a security perspective, I presume the user fills in the "Auth Token" text box with a token obtained from the signalk-generate-token utility ? (I know you can obtain tokens from performing a login using web sockets, but that is hardly what I would call user friendly), But the problem is that OpenCPN stores the token in clear text in the configuration file.

[image]

nohal, 2024-04-09:
So you desire the three dots from the button text to be removed, right?
OpenCPN is a locally run application and if the security of the user's account is compromised, their problems will likely be more severe than access to a Signal K token. But of course feel free to submit a pull request implementing safe encryption for OpenCPN's configuration data.

TwoCanPlugIn, 2024-04-09:
> So you desire the three dots from the button text to be removed, right?

Entirely up to you if you want to adhere to good UI guidelines. Today there are many instances where the UI is inconsistent.

Hakansv, 2024-04-09:
Today's master, Win 10
The new function to find SK-servers won't find any or list an address without a present SK-server.
I've have three local SignalK servers on these addresses (All port: 3000)
  192.168.1.73
  192.168.1.87
  192.168.1.154
Examples:
1.
Add new connection. Select Network SignalK
default address 0.0.0.0 : 10110
Click "Discover.."
-Not found
Empty address and port: - not found.

2.
I Entered a valid address where data is streaming and click "Discover...
- Not found.
Click again "Discover.. 
- He "found" an address: 192.168.1.167. There's no SK-server on that address!

BTW: UI note. What about some kind indication when the function is searching? "Busy spinner"??

nohal, 2024-04-09:
This functionality does a mDNS query, so something on 192.168.1.167 had to reply to that query saying it is a SK server. The query has nothing to do with the IP filled in the address field, it simply broadcasts on UDP port 5353 and waits if someone on the same subnet replies.
I would start by launching Wireshark and looking at what is going on on the network when you press the button.

Hakansv, 2024-04-09:
About SK on 192.168.1.167
Ok It's an old RPi that may have a SK installed but not initiated and wo any data.
But he ones listed the IP of my Win-PC where no SK-server is installed.
But the three working SK listed above was not found.

nohal, 2024-04-09:
Whoever replies to the query is filled in the address field. Wireshark will tell you how it looked on the network.

TwoCanPlugIn, 2024-04-09:
What is expected to happen if there are multiple SignalK Servers on the network, and there are multiple mDNS responses ?

BTW, Happy to see that SignalK traffic is now displayed in the debug window :-)

nohal, 2024-04-09:
Currently first one to respond wins. I am considering changing this to support multiple servers on the same network. Nobody ever considered it to be a use case out of labs.

PR #3758 (2024-04-08): Restore the Signal K server mDNS discovery
Restores the mDNS discovery of local Signal K server.
Tested on macOS and Linux
I have not restored and actually completely disabled even in the UI the option to discover the server dynamically at startup as it never was reliable enough anyway and made the startup take way too long in case the server was actually not available. I do not think there is much use case for it where the SK server has dynamic address that changes, but if someone sees any, am open to hear it.

Fixes #3757