OpenCPN/OpenCPN #3759: Windows bug

rgleason, 2024-03-26:
**Describe the bug**
See  screenshots please. Trying to build Windows. First tried Dave's recent branch, and it was failing even after cleaning and building all.
Then changed to master branch and git pulled, cleaned and tried building (several times) and get the same bug.

Here are some of the error screens.
[image]
[image]
[image]

Earlier ones were the same.

I was trying to learn how piano keys now work and the charts don't seem to be writing completely. The lower half of the screen often has a grey band.   Also OpenGL does not work.
[image]

**Desktop (please complete the following information if applicable):**
Windows 11 - OS: [e.g. macOS]
OpenCPN most current development  5.9

nohal, 2024-03-26:
Please confirm with latest CI package from https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

bdbcat, 2024-03-26:
Rick....
1.  There is no "Dave's recent branch".  What you saw me do on your AD session was to create a new branch locally on your PC, in case I needed to hack something.  No hack was needed, so the branch I created is no longer needed.  It exists only on your system.  Please delete it, to avoid confusion.  Master is where we are working now.
2. I see that you are not using GL mode.  You say GL mode does not work. Let us resolve that problem first, please.
3. I think there may be some problem with non-GL mode on very high resolution displays, as you have on your laptop.  We will address that separately.

nohal, 2024-03-26:
I would kind of expect "OpenGL not working" will turn out to be lack of `opencpn-glutil.exe` while running some local build product....

rgleason, 2024-03-26:
Ok Dave. Got it.    Where do I find opencpn-glutil.exe?
Pavel, I downloaded the current CI and piano keys are working great and the charts are working smoothly.  OpenGL works.

rgleason, 2024-03-26:
Maybe I'll find openglutil.exe in the one I just downloaded and copy it.

nohal, 2024-03-26:
Why do you do local build? Did you change some code? If not, save yourself the time and problems and simply use the CI packages. You are not losing anything, there is absolutely everything we did in them.

bdbcat, 2024-03-26:
If you want to copy openglutil.exe, look at the OCPN logfile after running your local build.  The log will tell you where it expected to find opnglutil.exe.  Just copy the .exe to that location, and all will be well.

rgleason, 2024-03-26:
I found it here
[image]

Log file

```
18:32:48.082 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
18:32:48.082 MESSAGE OCPNPlatform.cpp:714 OpenGL test utility not found at C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\bin\opencpn-glutil.exe.
18:32:48.082 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
18:32:48.082 MESSAGE OCPNPlatform.cpp:910 BuildGLCaps fails.
18:32:48.105 MESSAGE chcanv.cpp:2336 Metrics:  m_display_size_mm: 366     g_Platform->getDisplaySize():  3840:2400   
18:32:48.105 MESSAGE chcanv.cpp:2342 monitor size: 3840 2400
18:32:48.105 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: tcdata\harmonics-dwf-20210110-free.tcd
18:32:48.277 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: tcdata\HARMONICS_NO_US.IDX

```

This is blocking my opengl which is just fine and works well when allowed.

bdbcat, 2024-03-26:
18:32:48.082 MESSAGE OCPNPlatform.cpp:714 OpenGL test utility not found at C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\bin\opencpn-glutil.exe.

There is your problem.

nohal, 2024-03-27:
Rick, please, do everybody a favor and use the CI packages for your testing. We will explicitly tell you if anything else needs to be done, but your and everybody else's testing workflow is as simple as "Download from Cloudsmith, install and run".

rgleason, 2024-03-27:
I believe it is located right here
C:\Users\fcgle\source\repos\opencpn\build\glutil\RelWithDebInfo
and there is and exe and pdb.
Do I copy both files ? to 
C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\bin\opencpn-glutil.exe.

What can be done about dumb people like me, who think that MSvisualstudio will build?
I think CMAKE might need some added lines?  Or What am I doing wrong now?

Cleaned and rebuilt all.
43>------ Build started: Project: ALL_BUILD, Configuration: RelWithDebInfo Win32 ------
43>Building Custom Rule C:/Users/fcgle/source/repos/opencpn/CMakeLists.txt
========== Build: 43 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Build completed at 8:26 PM and took 02:28.840 minutes ==========

opencpn
Build started at 8:27 PM...
1>------ Build started: Project: i18n, Configuration: RelWithDebInfo Win32 ------
1>i18n: Done.
========== Build: 1 succeeded, 0 failed, 28 up-to-date, 0 skipped ==========
========== Build completed at 8:27 PM and took 00.756 seconds ==========

[image]

Thanks.
Rick

rgleason, 2024-03-27:
OK. I am not doing plugins right now. Will use the download.

rgleason, 2024-03-27:
I think it is still an issue.

nohal, 2024-03-27:
What is "it"?

bdbcat, 2024-03-27:
Do I copy both files ? to
C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\bin\opencpn-glutil.exe.

That is what the log suggests. That log message is there on purpose.  So just do it.

nohal, 2024-03-27:
If you follow the manual, you do not have to copy anything anywhere.

transmitterdan, 2024-03-27:
This patch to Opencpn\glutil\CMakeLists.txt will tell Visual Studio to do the copy step.
If not objection, I can submit a PR. This patch only affects local Visual Studio builds, not CI builds which are ok.
```
diff --git a/glutil/CMakeLists.txt b/glutil/CMakeLists.txt
index d50b72ad7..d9afbc29d 100644
--- a/glutil/CMakeLists.txt
+++ b/glutil/CMakeLists.txt
@@ -56,6 +56,22 @@ if (WIN32)
 else()
   install(TARGETS opencpn-glutil RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
 endif ()
+if (NOT OCPN_CI_BUILD)
+  if (MSVC)
+    add_custom_command(
+      TARGET opencpn-glutil
+      POST_BUILD
+      COMMAND ${CMAKE_COMMAND} -E copy_if_different "$<CONFIGURATION>/opencpn-glutil.exe"
+                "${CMAKE_BINARY_DIR}/$<CONFIGURATION>/opencpn-glutil.exe"
+    )
+    add_custom_command(
+      TARGET opencpn-glutil
+      POST_BUILD
+      COMMAND ${CMAKE_COMMAND} -E copy_if_different "$<CONFIGURATION>/opencpn-glutil.pdb"
+                "${CMAKE_BINARY_DIR}/$<CONFIGURATION>/opencpn-glutil.pdb"
+    )
+  endif (MSVC)
+endif (NOT OCPN_CI_BUILD)

 if (APPLE)
   install(
(
```

leamas, 2024-03-27:
hm... OCPN_CI_BUILD is, despite the name, not a flag necessarily tied to CI builds. We could fix this after issuing the PR to not clutter this "issue"

leamas, 2024-03-27:
> If you follow the manual, you do not have to copy anything anywhere.

This is the important message. If true, end of story. If not, we need to fix the developer manual and/or code.

nohal, 2024-03-27:
If I run `buildwin\winConfig.bat`, absolutely everything is done right, copied to where it should be and my development environment is ready to be used. Does anybody have other experience?

rgleason, 2024-03-27:
@bdbcat 

```
Do I copy both files ? to
C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\bin\opencpn-glutil.exe.

That is what the log suggests. That log message is there on purpose. So just do it.
```

That's what I did earlier.  Perhaps I was not clear about that.
[image]

I get the same  error
`PolyTriGroup *ppg = obj->pPolyTessGeo->Get_PolyTriGroup_head();`
`Unhandled exception at 0x77886859 (ntdll.dll) in opencpn.exe: 0xC000041D: An unhandled exception was encountered during a user callback.

OpenCPN.log file has towards the end:

```
08:58:38.354 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
08:58:38.354 MESSAGE OCPNPlatform.cpp:714 OpenGL test utility not found at C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\opencpn-glutil.exe.
08:58:38.354 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
08:58:38.354 MESSAGE OCPNPlatform.cpp:910 BuildGLCaps fails.
08:58:38.411 MESSAGE chcanv.cpp:2336 Metrics:  m_display_size_mm: 366     g_Platform->getDisplaySize():  3840:2400   
08:58:38.411 MESSAGE chcanv.cpp:2342 monitor size: 3840 2400
```

So now it doesn't want the files in "bin"  it wants them in the directory above "RelWithDebInfo"?
But I think it found the file anyway   "908 OpenGL test complete."

Why is  "BuildGLCaps failing?"  What is it?

rgleason, 2024-03-27:
I copied the opencpn-glutil  files to  RelWithDebInfo directory.

[image]

This is the Log result now. But it still fails.

```
09:17:48.339 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
09:17:48.340 MESSAGE OCPNPlatform.cpp:722 Starting OpenGL test utility: C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\opencpn-glutil.exe opengl-info C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\gl_caps.json
09:17:50.357 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
09:17:50.357 MESSAGE OCPNPlatform.cpp:930 OpenGL determined CAPABLE.
```

rgleason, 2024-03-27:
Restarted MSstudio twice and rebuilt again. Now its working with OpenGL.
Now closing this, but I just question why the opencpn-glutil.exe and pdb aren't located where OpenCPN can use them to check opengl ?

rgleason, 2024-03-27:
I copied the opencpn-glutil  files to  RelWithDebInfo directory.

This is the Log result now. But it still fails.

```
09:17:48.339 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
09:17:48.340 MESSAGE OCPNPlatform.cpp:722 Starting OpenGL test utility: C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\opencpn-glutil.exe opengl-info C:\Users\fcgle\source\repos\opencpn\build\RelWithDebInfo\gl_caps.json
09:17:50.357 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
09:17:50.357 MESSAGE OCPNPlatform.cpp:930 OpenGL determined CAPABLE.
```

closed 2024-03-27 by rgleason

rgleason, 2024-03-28:
I like this  by TDan https://github.com/OpenCPN/OpenCPN/issues/3759#issuecomment-2021791310
Note, I do not use the wxWidgets setup that is normally used.

transmitterdan, 2024-03-28:
The INSTALL project copies the glutil artifacts to the installation folder. But within Visual Studio, most people don't build the INSTALL project.  They build the 'opencpn' project because it is the 'startup' project.  That project does not copy the glutil artifacts anywhere. They are left in their original build folder locations. Years ago, I added the post-build-event methods to the cmake process to copy plugin dlls and pdbs to the correct local runtime folder.  I will submit PR shortly to do the same for glutil artifacts.  That should be a final solution.  It should have no affect on CI builds. The CI builds of the PR will tell us if that is so.

Hakansv, 2024-03-28:
FYI: I build the INSTALL project by VS2022 and NSIS set up works fine. Just remember to all time use "rebuild project INSTALL" otherwise possible changes to inbound plugins are not included.

rgleason, 2024-03-28:
Thanks TDan
https://github.com/OpenCPN/OpenCPN/pull/3767/commits/8f124a322eedf150eff1cb5e14eed0a2359b333d