OpenCPN/OpenCPN #3760: Avoid log flooding of "Render texture"

Hakansv, 2024-03-27:
The log file is sometimes "flooded" by equal Render texture messages. See below.
I assume the message is there for a reason but it would be enough to message when changed?
If no objections or other views I'll make a PR to log only when the value is changed.

```
08:32:02.382 MESSAGE RolloverWin.cpp:189 Render texture  108
08:32:10.694 MESSAGE RolloverWin.cpp:189 Render texture  108
08:32:10.706 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:10.930 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:11.426 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:11.926 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:12.429 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:12.927 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:13.432 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:13.926 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:14.427 MESSAGE RolloverWin.cpp:189 Render texture  133
08:32:14.930 MESSAGE RolloverWin.cpp:189 Render texture  133
08:33:09.512 MESSAGE RolloverWin.cpp:189 Render texture  1101
08:33:10.102 MESSAGE RolloverWin.cpp:189 Render texture  1101
```

leamas, 2024-03-27:
IMHO, better to just remove this debug message which as far as I tell see does not belong to anything in production.

If anyone actually wants it to stay  it should be a wxLogDebug() instead to avoid it in normal logs.

leamas, 2024-03-27:
Sidenote: https://docs.wxwidgets.org/3.0/classwx_log.html#ae814f9b9b4c8b4a52e34d7c919f98296

Hakansv, 2024-03-27:
Leamas will fix

closed 2024-03-27 by Hakansv