OpenCPN/OpenCPN #3763: Plugin API PlugInPlaySoundEx(soundFile) does not play sound if space in soundFile string

antipole2, 2024-03-27:
**Describe the bug**
As in the title.
On MacOS, spaces in file names are common.

**To Reproduce**
Steps to reproduce the behaviour:
1. Name the OCPN app with a space - say 'OpenCPN v5.9'
2. Launch it
3. Do something that would play a sound - ship's bells?
4. no sound
5. Rename the app to, say, 'OpenCPNv5.9'
6. Relaunch
7. Sound plays OK.

Issue arises because the file string to the sound file includes the app name.  If there is a space in it, it does not play.
**Expected behaviour**
Should play sound.

**Desktop**
 - MacOS Sonoma
 - OCPN v5.8 and v5.9

antipole2, 2024-07-02:
Still the case in Beta 2

leamas, 2024-07-08:
@antipole2  Can perhaps fix with some help.  Could you please provoke the error while using --loglevel debug and grep for *Sound command:* in the log file and report back.

leamas, 2024-07-09:
@antipole2 : I have test builds  available at https://cloudsmith.io/~alec-leamas/repos/opencpn/packages/, look for the 3781321b2 builds. This is based on the assumption that the problem stems from missing argument quoting in the  CLI command string.

You can also build yourself,  attaching patch. In any case,  I need you to test and verify in order to be able to merge this.

[0001-Sounds-Add-quoting-of-cli-arguments-3763.patch.gz](https://github.com/user-attachments/files/16145565/0001-Sounds-Add-quoting-of-cli-arguments-3763.patch.gz)

antipole2, 2024-07-09:
Alec - thanks for working on this.  I would test it but am on a remote anchorage in Sweden (Bohuslän coast) with only 1 bar mobile signal.  Likely some days before I can download a build.
To test, you need to edit the name of the application to include a space, say 'OpenCPN v5.9'

leamas, 2024-07-10:
> To test, you need to edit the name of the application to include a space, say 'OpenCPN v5.9'

Easier said than done -- I have no Mac available.

@nohal : have you any possibility to make a check so we perhaps could be able to close this?

leamas, 2024-07-10:
...but then again, the basic problem of OpenCPN configuring tz and formatting in multiple places is a real one.

antipole2, 2024-07-22:
Looks like this kicked own the road but still mile-stoned for v5.10

leamas, 2024-07-22:
I should not trust the milestone that much.  It's basically a best effort undertaking.

Welcome to Sweden, to the parts I once was born in! 

That said, as I see it the ball is now on your side: you reported the issue, have the hardware ant the test case. I can make a new build when you are ready to test, the old is most likely gone,  but that is basically it.

bdbcat, 2025-04-26:
Closing.
Please re-open if necessary for o5.12

closed 2025-04-26 by bdbcat