OpenCPN/OpenCPN #3766: RFE  Chart Hamburger - Display Options

rgleason, 2024-03-27:
**Is your feature request related to a problem? Please describe.**
Add to "Display Options"

- Show Chart Bar.
- Show Extended Chart Bar Information.

Remove Enable Chart Bar Extended from Display Advanced.
For consistency.

rgleason, 2024-04-01:
No. I am not going to do that. One topic per issue.
I think you should hide or remove your comment.

leamas, 2024-04-01:
this  is issue is still incomprehensible. Please answer to standard Enhancement questions  fron the form above or close.

rgleason, 2024-04-01:
Maybe to you, but not to others.

nohal, 2024-04-01:
FWIW I also have absolutely no idea what this talks about, so our list of people that can do something about it starts to get kind of short...

rgleason, 2024-04-01:
Dave understands it.

leamas, 2024-04-01:
>  Hopefully it is clearer now.

No.

bdbcat, 2024-04-01:
Clear to me.  Considering....

rgleason, 2024-05-07:
Close?