OpenCPN/OpenCPN #3769: RFE: New User: Make O-charts_pi an internal plugin

rgleason, 2024-03-29:
**Is your feature request related to a problem? Please describe.**
New user request: Make O-charts and internal plugin so it does not have to be 
Downloaded and Installed.

Quite frankly, I don't see what the problem is. It is downloaded easily by plugin manager, installed and enabled.
What is the advantage of making it internal?
What are the problems with making it internal?  <--Then it looks like OpenCPN and o-charts are the same, which they aren't.

transmitterdan, 2024-03-30:
Rick,

It's impossible to have a "brief quick start guide" work for every user.  Different brains are wired differently. Some brains are expecting one thing and other brains a different thing.  Some people have differing levels of patience and different logic pathways.  Without significant training in behavioral psychology, it's impossible to anticipate the neural pathways that some outlier personality types will take.

We should not make the O-charts plugin internal.  The release cycle for O and O-charts would have to be synchronized which will lead to more problems than it solves.

Frankly, it will not help the helpless user to pre-install O-charts and a single (free?) chart.  They will just get mad trying to add another chart.  The training that is forced on a new user to get their first chart visible is not a bad thing.  It's actually a good thing.  For those without the ability to make it that far without screaming in agony, maybe O just isn't for them.  We have to face the fact that not everyone is going to like the product.  As long as that's a suitably small percentage (ignoring their ability to yell loudly) then we are doing ok. But we should not change what is working for 99% of users to accommodate the 1% of outlier brain types regardless of how much noise the 1% make.  All we can do, is help the 1% get over the initial learning curve as best we can.

bdbcat, 2024-03-30:
What he said....

Hakansv, 2024-03-30:
I do appreciate T-Dan's conclusions.

rgleason, 2024-03-30:
I dont think wontfix is right. But wontdo is. Closing.

rgleason, 2024-03-30:
There is also  "cantdo".  Which is another limit. There are some thoughts out there about "developer" heads that "don't get it", its always good to keep some perspective, sometimes its difficult we are so close to it, but I think TDan is right, we need to force some training on new users by downloading with Chartdownloader. 
Closing.

closed 2024-03-30 by rgleason