OpenCPN/OpenCPN #3772: Dashboard Wind History text is inconsistent/wrong

transmitterdan, 2024-04-01:
**Describe the bug**
Wind history graph has unreadable text.  Font too big/wrong position.

[image]

**Desktop (please complete the following information if applicable):**
 - OS: all
 - Version current master

Hakansv, 2024-04-01:
I suppose it was meant to be shown wider than your shot when it was created many years ago. That's anyway how I use it. 
The first and simple workaround would be to increase the default width to give space for all text.
Font sizes may as many fonts nowadays be better adapted to scaled display. Agree. 
The statistics could be drawn on a new row but I think the idea was to shrink the height of the instrument as much as possible?
Suggestions?

[image]

Hakansv, 2024-04-01:
Personally I think the statistics are of minor interest. We could maybe try to not print that text if not space enough?

BerndCirotzki, 2024-04-01:
I have to get involved sometimes, after all I was tinkering with the plugin.
The first optic was always like that, and you had to move the window. But it is now possible to build the dashboard "once" in the distant future, just as you want it to be.

What I noticed again here is that the window title "for my taste" is too large and takes up too much space on the screen when there are several dashboards.
That's why I built my own dashboard, which I always replace for myself. But I'm of the opinion that this doesn't necessarily make sense for a long time.
That's why I just made my code public. https://github.com/BerndCirotzki
The problem is that I only work with Windows and therefore I can't test it on other systems and make a "pull request" to OpenCPN. But maybe one of you is interested and finds that
possibly good too, and extends it for other operating systems.
- However, the change to "dockable" was only possible by completely removing the AUI manager.
- I completely integrated the engine dashboard because I wanted to have “color” and “dockabel” there too.

... as I said, it's a matter of taste.

bc38, 2024-04-02:
> I have to get involved sometimes, after all I was tinkering with the plugin. The first optic was always like that, and you had to move the window. But it is now possible to build the dashboard [...]

are you planning to propose these changes to the general dashboard plugin branch?
in particular engine, I think that everything related to the dashboard should be within one plugin

Hakansv, 2024-04-02:
@BerndCirotzki  You have a sophisticated change idea but since I'm none of the "competent "O" developers " I'll just make a try to fix the bug reported by:
@transmitterdan 
I've tried to draw the clashing texts in a another way. It also not draw the statistics if not enough space. See picts.
If you think that will solve this bug I'll try a PR?

Minimum default width:
[image]

Somewhat widened:
[image]

closed 2024-04-06 by transmitterdan