OpenCPN/OpenCPN #3774: 5.9.0  does not install on M1 macMini

bcn2, 2024-04-01:
Installing the latest versions of 5.9.0on our M1 (14.2.1) will fail. Tested with 1376 and 1377 from today
[image]

Btw.: what is the meaning of "11" and "12" in the pgk names?

nohal, 2024-04-01:
The correct package to install is https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-universal+8544.f4d68bd2a-14.pkg and works just fine for me.

We should probably not even upload the ones from GA to Cloudsmith, they serve just to test if we still build with Homebrew on all the different macOS versions.

bcn2, 2024-04-01:
That one does crash at start-up:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenCPN [95473]
Path:                  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.9.0-universal+f4d68bd2a ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-04-01 15:03:16.2954 +0200
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        24089CE9-363A-CCBA-FEA0-0B62E0E783C4

Time Awake Since Boot: 2200000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000001e0
Exception Codes:       0x0000000000000001, 0x00000000000001e0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [95473]

VM Region Info: 0x1e0 is not in any region.  Bytes before following region: 68719476256
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libGL.dylib                   	       0x1ee731f50 glDeleteTextures + 16
1   OpenCPN                       	       0x100bb1398 RoutePoint::~RoutePoint() + 144
2   OpenCPN                       	       0x100bb1508 RoutePoint::~RoutePoint() + 12
3   OpenCPN                       	       0x100b490c0 GPXLoadRoute1(pugi::xml_node&, bool, bool, bool, int, bool, bool) + 3296
4   OpenCPN                       	       0x100b4e450 NavObjectCollection1::LoadAllGPXObjects(bool, int&, bool) + 308
5   OpenCPN                       	       0x100858bb4 MyConfig::LoadNavObjects() + 3312
6   OpenCPN                       	       0x10088ca28 MyFrame::OnInitTimer(wxTimerEvent&) + 4048
7   libwx_baseu-3.2.0.2.3.dylib   	       0x101fdf370 wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 376
8   libwx_baseu-3.2.0.2.3.dylib   	       0x101fe00c8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 120
9   libwx_baseu-3.2.0.2.3.dylib   	       0x101fdff28 wxEvtHandler::ProcessEvent(wxEvent&) + 56
10  libwx_baseu-3.2.0.2.3.dylib   	       0x101fe0418 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
11  libwx_baseu-3.2.0.2.3.dylib   	       0x101f95d80 wxTimerImpl::SendEvent() + 100
12  CoreFoundation                	       0x185f5ba20 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32
13  CoreFoundation                	       0x185f5b6c8 __CFRunLoopDoTimer + 972
14  CoreFoundation                	       0x185f5b200 __CFRunLoopDoTimers + 356
15  CoreFoundation                	       0x185f3e744 __CFRunLoopRun + 1856
16  CoreFoundation                	       0x185f3d9ac CFRunLoopRunSpecific + 608
17  HIToolbox                     	       0x1904ec448 RunCurrentEventLoopInMode + 292
18  HIToolbox                     	       0x1904ec284 ReceiveNextEventCommon + 648
19  HIToolbox                     	       0x1904ebfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
20  AppKit                        	       0x18971a8a4 _DPSNextEvent + 660
21  AppKit                        	       0x189ef4980 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
22  AppKit                        	       0x18970dd50 -[NSApplication run] + 476
23  libwx_osx_cocoau_core-3.2.0.2.3.dylib	       0x1029cf5e8 wxGUIEventLoop::OSXDoRun() + 140
24  libwx_baseu-3.2.0.2.3.dylib   	       0x101fc6340 wxCFEventLoop::DoRun() + 40
25  libwx_baseu-3.2.0.2.3.dylib   	       0x101f3d398 wxEventLoopBase::Run() + 120
26  libwx_baseu-3.2.0.2.3.dylib   	       0x101f18228 wxAppConsoleBase::MainLoop() + 120
27  libwx_baseu-3.2.0.2.3.dylib   	       0x101f65718 wxEntry(int&, wchar_t**) + 104
28  OpenCPN                       	       0x100871848 main + 24
29  dyld

nohal, 2024-04-01:
Not for me, will try to reproduce, but it certainly has nothing to do with the packaging.

bcn2, 2024-04-01:
Tried with plug-ins disabled -same result.

nohal, 2024-04-01:
Yes, nothing to do with plugins, it crashes when loading `navobj.xml`

bdbcat, 2024-04-01:
I have seen this.  Nothing wrong with navobj,  The fault I think occurs for me when glutil is not found, so glDeleteTexture is failing. 
Need to inspect logfile, please.

bcn2, 2024-04-01:
I do see indeed a glimpse of the raster chart before the crash...

5:08:29.021    INFO logger.cpp:130 ------- OpenCPN version 5.9.0-universal+f4d68bd2a restarted at 2024-04-01 -------

15:08:29.021    INFO logger.cpp:130 Using loglevel INFO
15:08:29.021 MESSAGE ocpn_app.cpp:1143 wxWidgets version: wxWidgets 3.2.5 Apple Mac OS X 64 bit wxMac
15:08:29.021 MESSAGE ocpn_app.cpp:1148 OS Version reports as:  14.2
15:08:29.021 MESSAGE ocpn_app.cpp:1150 MemoryStatus:  mem_total: 8192 mb,  mem_initial: 84 mb
15:08:29.021 MESSAGE ocpn_app.cpp:1162 OCPN_OSDetail:  arm64 ; darwin-wx32 ; 14.1 ;  ; 
15:08:29.022 MESSAGE ocpn_app.cpp:1169 SData_Locn is /Applications/OpenCPN.app/Contents/SharedSupport/
15:08:29.022 MESSAGE ocpn_app.cpp:1184 PrivateDataDir is /Users/renate/Library/Preferences/opencpn
15:08:29.022 MESSAGE ocpn_app.cpp:1250 Using existing Config_File: /Users/renate/Library/Preferences/opencpn/opencpn.ini
15:08:29.028 MESSAGE navutil.cpp:1229 winPluginDir, read from ini file: 
15:08:29.028 MESSAGE navutil.cpp:1311 Setting Ownship Lat/Lon 41.2752, 1.9662
15:08:29.044 MESSAGE navutil.cpp:3261 Switch InlandEcdis mode Off
15:08:29.044 MESSAGE ConfigMgr.cpp:544 Loading Configs catalog: /Users/renate/Library/Preferences/opencpn/Configs/configs.xml
15:08:29.046 MESSAGE styles.cpp:841 Styles loading from /Applications/OpenCPN.app/Contents/SharedSupport/uidata/styles.xml
15:08:29.048 MESSAGE styles.cpp:818 No styles found at: /Users/renate/Library/Preferences/opencpn/
15:08:29.048 MESSAGE styles.cpp:801 No styles found at: /Users/renate/Library/Preferences/opencpn/.opencpn/
15:08:29.060 MESSAGE styles.cpp:775 Using SVG Icons
15:08:29.060 MESSAGE ocpn_app.cpp:1311 Detected display size (horizontal): 878 mm
15:08:29.060 MESSAGE OCPNPlatform.cpp:1008 Getting DefaultSystemLocale...
15:08:29.061 MESSAGE ocpn_app.cpp:1347 System default Language:  es_ES
15:08:29.061 MESSAGE ocpn_app.cpp:1350 Config file language:  en_US
15:08:29.061 MESSAGE ocpn_app.cpp:1355 Adjusted App language:  en_US
15:08:29.061 MESSAGE OCPNPlatform.cpp:1089 ChangeLocale: Language load for:  en_US
15:08:29.071 MESSAGE OCPNPlatform.cpp:1142 ChangeLocale: Locale Init OK for:  en_US
15:08:29.071 MESSAGE OCPNPlatform.cpp:1167 Loading catalog for opencpn core.
15:08:29.074 MESSAGE OCPNPlatform.cpp:1008 Getting DefaultSystemLocale...
15:08:29.074 MESSAGE ocpn_app.cpp:1362 Opencpn language set to:  en_US
15:08:29.075 MESSAGE ocpn_app.cpp:1581 Creating MyFrame...size(2637, 1497)  position(764, 80)
15:08:29.094 MESSAGE ocpn_frame.cpp:734  **** Connect stuff
15:08:29.094 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
15:08:29.095 MESSAGE OCPNPlatform.cpp:722 Starting OpenGL test utility: /Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil opengl-info /Users/renate/Library/Preferences/opencpn/gl_caps.json
15:08:29.102 MESSAGE OCPNPlatform.cpp:726 OpenGL test utility failed with exit code -6
15:08:29.102 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
15:08:29.102 MESSAGE OCPNPlatform.cpp:910 BuildGLCaps fails.
15:08:29.114 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /Applications/OpenCPN.app/Contents/SharedSupport/tcdata/harmonics-dwf-20210110-free.tcd
15:08:29.144 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /Applications/OpenCPN.app/Contents/SharedSupport/tcdata/HARMONICS_NO_US.IDX
15:08:29.153 MESSAGE chcanv.cpp:2310 Metrics:  m_display_size_mm: 878     g_Platform->getDisplaySize():  3840:1600   
15:08:29.153 MESSAGE chcanv.cpp:2316 monitor size: 3840 1600
15:08:29.157 MESSAGE chcanv.cpp:2310 Metrics:  m_display_size_mm: 878     g_Platform->getDisplaySize():  3840:1600   
15:08:29.157 MESSAGE chcanv.cpp:2316 monitor size: 3840 1600
15:08:29.162 MESSAGE chartsymbols.cpp:699 ChartSymbols loaded from /Applications/OpenCPN.app/Contents/SharedSupport/s57data/chartsymbols.xml
15:08:29.173 MESSAGE ocpn_frame.cpp:8371 Using s57data in /Applications/OpenCPN.app/Contents/SharedSupport/s57data
15:08:29.246 MESSAGE chartdb.cpp:219 ChartDB Cache policy:  Max open chart limit is 20.
15:08:29.247 MESSAGE chartdbs.cpp:156 Loading chart db version: V018
15:08:29.247 MESSAGE chartdbs.cpp:1195 Chartdb: Chart directory list follows
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #0: /Users/renate/Documents/Charts/ENC/US_REGION04
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #1: /Users/renate/Documents/Charts/oeuSENC-ITe
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #2: /Users/renate/Documents/Charts/ENC/Elbe_2020-12-17/ENC_ROOT
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #3: /Users/renate/Documents/Charts/ENC/FR_INLAND
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #4: /Users/renate/Documents/Charts/ENC/NL_INLAND
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #5: /Users/renate/Documents/Charts/ENC/US_REGION24
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #6: /Users/renate/Library/Preferences/opencpn/s63/s63charts/AU
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #7: /Users/renate/Documents/Charts/oeuSENC-CU
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #8: /Users/renate/Documents/Charts/ENC/oeuSENC-FRmed
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #9: /Users/renate/Documents/Charts/ENC/oeuSENC-BE_NL
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #10: /Users/renate/Documents/Charts/oeuSENC-GR
15:08:29.247 MESSAGE chartdbs.cpp:1214   Chart directory #11: /Users/renate/Documents/Charts/oeuSENC-PAc
[... cut ...]
15:08:29.285 MESSAGE chartdb.cpp:1284 Initializing Chart /Users/renate/Documents/Charts/RNC/IMRAY/C70-2018.kap
15:08:29.287 MESSAGE chartimg.cpp:5025    VP Final Check: Georeference Chart_Error_Factor on chart /Users/renate/Documents/Charts/RNC/IMRAY/C70-2018.kap is 0.00229346 
     nominal pixel error is: 7.00693
15:08:29.287 MESSAGE ocpn_app.cpp:1828 OpenCPN Initialized in 4811 ms.
15:08:30.058 MESSAGE rest_server.cpp:400 Listening on http://0.0.0.0:8443

15:08:30.130 MESSAGE ocpn_frame.cpp:4639 OnInitTimer...0

bdbcat, 2024-04-01:
There it is:
15:08:29.095 MESSAGE OCPNPlatform.cpp:722 Starting OpenGL test utility: /Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil opengl-info /Users/renate/Library/Preferences/opencpn/gl_caps.json
15:08:29.102 MESSAGE OCPNPlatform.cpp:726 OpenGL test utility failed with exit code -6

nohal, 2024-04-01:
But this should simply disable OpenGL, not lead to a crash.

bdbcat, 2024-04-01:
Above.  I got that part.

nohal, 2024-04-01:
I just tested with `ocpn-glutil` missing, does not crash at startup, but does crash exactly on the same spot on exit.

bcn2, 2024-04-01:
Not seen issues with OpenGL on that system in the past.....

bdbcat, 2024-04-01:
Compound problem. Something fishy about the GL detect/test utility, as well as fault in no-GL mode.

nohal, 2024-04-01:
Simple test - open Terminal and run `/Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil opengl-info`

bcn2, 2024-04-01:
Mac-mini:~ renate$ /Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil opengl-info
dyld[96262]: Library not loaded: /usr/local/lib/libwx_osx_cocoau_gl-3.2.dylib
  Referenced from: <382D1C12-3C65-3CA4-BE4E-9E1DD5D84E32> /Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil
  Reason: tried: '/usr/local/lib/libwx_osx_cocoau_gl-3.2.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/libwx_osx_cocoau_gl-3.2.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau_gl-3.2.dylib' (no such file)
Abort trap: 6
Mac-mini:~ renate$

nohal, 2024-04-01:
OK, this part of the problem is clear, will take care of it

bdbcat, 2024-04-01:
I fixed the crash on no-GL. 
MacOS on retina display has lots of trouble in no-GL mode.  Scaling issues, mainly.  But also, more severe, it crashes when shifting **out** of no-GL mode, before updating the config file.  So stuck forever, without manual config file edit.
Lets leave this issue open for that reason only.

bcn2, 2024-04-01:
With 8594.1ecd4b794 still crashing....

bcn2, 2024-04-01:
Start without plug-ins does work, but there is no way to activate plug-ins
[image]

bdbcat, 2024-04-01:
Are you actually in GL mode at startup now?

bcn2, 2024-04-01:
In the GUI OpenGL shows up as active.

22:05:10.599 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
22:05:10.599 MESSAGE OCPNPlatform.cpp:722 Starting OpenGL test utility: /Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil opengl-info /Users/renate/Library/Preferences/opencpn/gl_caps.json
22:05:10.713 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
22:05:10.713 MESSAGE OCPNPlatform.cpp:930 OpenGL determined CAPABLE.
22:05:10.723 MESSAGE chcanv.cpp:933 Creating glChartCanvas

Run was started in "Safe Mode" with plug-ins as shown in the screen-shot.

Closing and running again results in crash.

nohal, 2024-04-01:
If the crash report is generated, we want to see it. If started with OpenGL enabled, everything works just fine for me.

bcn2, 2024-04-01:
the log

22:14:39.944 MESSAGE ocpn_frame.cpp:734  **** Connect stuff
22:14:39.944 MESSAGE OCPNPlatform.cpp:901 Starting OpenGL test...
22:14:39.944 MESSAGE OCPNPlatform.cpp:722 Starting OpenGL test utility: /Applications/OpenCPN.app/Contents/MacOS/opencpn-glutil opengl-info /Users/renate/Library/Preferences/opencpn/gl_caps.json
22:14:40.070 MESSAGE OCPNPlatform.cpp:908 OpenGL test complete.
22:14:40.070 MESSAGE OCPNPlatform.cpp:930 OpenGL determined CAPABLE.
22:14:40.080 MESSAGE chcanv.cpp:933 Creating glChartCanvas

and the crash report
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenCPN [98101]
Path:                  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.9.0-universal+1ecd4b794 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-04-01 22:13:14.4475 +0200
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        24089CE9-363A-CCBA-FEA0-0B62E0E783C4

Time Awake Since Boot: 2200000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000300
Exception Codes:       0x0000000000000001, 0x0000000000000300

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [98101]

VM Region Info: 0x300 is not in any region.  Bytes before following region: 4302847232
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100784000-100fc4000    [ 8448K] r-x/r-x SM=COW  ...MacOS/OpenCPN

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   OpenCPN                       	       0x100c3a630 PluginLoader::LoadPluginCandidate(wxString const&, bool) + 2856
1   OpenCPN                       	       0x100c39030 PluginLoader::LoadPlugInDirectory(wxString const&, bool) + 1444
2   OpenCPN                       	       0x100c37e28 PluginLoader::LoadAllPlugIns(bool, bool) + 5552
3   OpenCPN                       	       0x1009382f4 MyFrame::OnInitTimer(wxTimerEvent&) + 1116
4   libwx_baseu-3.2.0.2.3.dylib   	       0x10208b370 wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 376
5   libwx_baseu-3.2.0.2.3.dylib   	       0x10208c0c8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 120
6   libwx_baseu-3.2.0.2.3.dylib   	       0x10208bf28 wxEvtHandler::ProcessEvent(wxEvent&) + 56
7   libwx_baseu-3.2.0.2.3.dylib   	       0x10208c418 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
8   libwx_baseu-3.2.0.2.3.dylib   	       0x102041d80 wxTimerImpl::SendEvent() + 100
9   CoreFoundation                	       0x185f5ba20 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32
10  CoreFoundation                	       0x185f5b6c8 __CFRunLoopDoTimer + 972
11  CoreFoundation                	       0x185f5b200 __CFRunLoopDoTimers + 356
12  CoreFoundation                	       0x185f3e744 __CFRunLoopRun + 1856
13  CoreFoundation                	       0x185f3d9ac CFRunLoopRunSpecific + 608
14  HIToolbox                     	       0x1904ec448 RunCurrentEventLoopInMode + 292
15  HIToolbox                     	       0x1904ec284 ReceiveNextEventCommon + 648
16  HIToolbox                     	       0x1904ebfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
17  AppKit                        	       0x18971a8a4 _DPSNextEvent + 660
18  AppKit                        	       0x189ef4980 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
19  AppKit                        	       0x18970dd50 -[NSApplication run] + 476
20  libwx_osx_cocoau_core-3.2.0.2.3.dylib	       0x102a7b5e8 wxGUIEventLoop::OSXDoRun() + 140
21  libwx_baseu-3.2.0.2.3.dylib   	       0x102072340 wxCFEventLoop::DoRun() + 40
22  libwx_baseu-3.2.0.2.3.dylib   	       0x101fe9398 wxEventLoopBase::Run() + 120
23  libwx_baseu-3.2.0.2.3.dylib   	       0x101fc4228 wxAppConsoleBase::MainLoop() + 120
24  libwx_baseu-3.2.0.2.3.dylib   	       0x102011718 wxEntry(int&, wchar_t**) + 104
25  OpenCPN                       	       0x10091dc78 main + 24
26  dyld                          	       0x185ae10e0 start + 2360

bdbcat, 2024-04-01:
Quick hint:
If you are in GL mode, the main toolbar has rounded corners.  If in No-GL mode, the toolbar has square corners.

bcn2, 2024-04-01:
In the ini file:
[PlugIns/libobjsearch_pi.dylib]
bEnabled=0
[PlugIns/OBJSEARCH]
CloseOnShow=1
LimitRange=0

nohal, 2024-04-01:
Objsearch is not a problem for me, at least no the current version form the catalog.

bdbcat, 2024-04-01:
The rounded corners hint:  Made that so we can see from a screenshot what mode a user is in, without needing to ask for a log.

bcn2, 2024-04-01:
Some more plug-ins uninstalled,  however the crash persists.
The next plug-in in the ini after objectsearch was meteovache. Manually disabled. No difference.

The Options|plug-in tab does not allow to select and edit plug-ins. No plug-in candidates listed neither.
[image]

bcn2, 2024-04-01:
[image]

nohal, 2024-04-01:
Remember that you are now running natively on ARM, so any plugins built for x86 are incompatible. Any plugins in the catalog not listed as available for ARM will not be shown neither.
But the list certainly should not be empty as on your picture, there already are plugins built for ARM in the master catalog.

bcn2, 2024-04-02:
Which architecture, please?
[image]

bcn2, 2024-04-02:
After going back to 5.8.4 (installed in parallel) and forth to 5.9.0 8598 - the latest one - all is back to normal. Some plug-ins not in catalog, but the rest ok.

bdbcat, 2024-04-04:
Did we ever figure out the root cause of the Plugin crashes and PIM problems?
Probably something to do with multiple versions of OCPN, and incompatible architecture of plugins introduced by MacOS universal builds
In any event, the origin issue from the top is now well understood, and corrected.
So, close?

nohal, 2024-04-04:
Yes, this particular issue is solved.

bdbcat, 2024-04-10:
BCN...
Please close, if satisfied.

closed 2024-04-10 by bcn2