OpenCPN/OpenCPN #3778: Track properties - Width px setting not honored

bcn2, 2024-04-02:
5.9.0, track properties

right click, changing the pixel width parameter does not modify the drawn track.

nohal, 2024-04-02:
Seems working fine for me. Aren't you using the default yellow color with the yellow highlight on?

bcn2, 2024-04-02:
Default - but as it results too bolt, trying to narrow.
[image]

nohal, 2024-04-02:
Change the color of the track or
Ships -> OwnShip -> Higihlight track - uncheck the checkbox

bcn2, 2024-04-03:
That was it - after changing the setting, thre was not contextual menu for tracks and as well from the Routemanager track propertities did not open. After a restart of OpenCPN ok.

closed 2024-04-03 by bcn2