OpenCPN/OpenCPN #3779: Printing a chart shouldn't have toolbars on it

RooieDirk, 2024-04-03:
The new toolbar is always shown on a (paper)printout. To my opinion a print should be without toolbars, if I wanted a picture including toolbars I do a printscreen. Same for the canvas-toolbar.
The gps-compass and chartbar are also shown, but at least I can hide them if I want. Picture with a print preview:
[image]

bdbcat, 2024-04-03:
Well, yeah, sure.
Thanks for the catch,  should be an easy update.

RooieDirk, 2024-04-03:
I already had a look myself, but the toolbar doesn't seem tohave a show(bool) function or something similar. (As far as I could see)

bdbcat, 2024-04-03:
Nope, it doesn't.  the toolbar is THE UI. Consider:  Turn off the toolbar.  How would one access any UI to turn the toolbar back on?

RooieDirk, 2024-04-04:
Found it myself ;-) Pull request posted.

bdbcat, 2024-04-04:
Nice.
Merged.

RooieDirk, 2024-04-04:
Forget something. There is also a thing called glChartCanvas. I tried already doing the same trick there, but this doesn't work. Looks as if a Refresh() doesn't apply to glChartCanvas.

Hakansv, 2024-04-05:
On windows neither single or multi canvas are working. The tool and menu bars are still on print.
(On multi canvas only left canvas is printed.)

[image]

RooieDirk, 2024-04-05:
Ha Hakan, 
On current master it should work if you don't use GL.

Hakansv, 2024-04-05:
Hmm - Most users does more or less mandatory

nohal, 2024-04-08:
We probably still want to get rid of the rest of the overlay artifacts

[image]

RooieDirk, 2024-04-09:
Yes, but the other one's  an user is able to toggle in the options. And there might be different wishes for different users. Grid yes/no, Scalebar y/n, depthunits y/n
For chartbar, gpscompas and even ownship I don't see a user case om paper.
btw I am working on a PI to add chartborders, so a paper print would be really usable as backup navigation chart.
[image]

nohal, 2024-04-09:
Exactly that is what I meant, the chart piano, the embossed text and the GPS status indicator should not be there.

closed 2024-11-22 by RooieDirk