OpenCPN/OpenCPN #3782: Build fails with LTO

eli-schwartz, 2024-04-04:
I tried to build with the following *FLAGS to optimize the build: `-flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

Note the -Werror=* flags are used to help detect cases where the compiler tries to optimize by assuming UB cannot exist in the source code -- if it does exist, ordinarily the code would be miscompiled, and this says to make the miscompilation a fatal error.

I got this error:
```
FAILED: opencpn 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -pthread -D__WXGTK3__ -Wl,-O1 -Wl,--as-needed -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0   -Wl,--export-dynamic  -rdynamic CMakeFiles/opencpn.dir/src/about.cpp.o CMakeFiles/opencpn.dir/src/AboutFrame.cpp.o CMakeFiles/opencpn.dir/src/AboutFrameImpl.cpp.o CMakeFiles/opencpn.dir/src/ais.cpp.o CMakeFiles/opencpn.dir/src/AISTargetAlertDialog.cpp.o CMakeFiles/opencpn.dir/src/AISTargetListDialog.cpp.o CMakeFiles/opencpn.dir/src/AISTargetQueryDialog.cpp.o CMakeFiles/opencpn.dir/src/ais_info_gui.cpp.o CMakeFiles/opencpn.dir/src/CanvasConfig.cpp.o CMakeFiles/opencpn.dir/src/canvasMenu.cpp.o CMakeFiles/opencpn.dir/src/CanvasOptions.cpp.o CMakeFiles/opencpn.dir/src/catalog_mgr.cpp.o CMakeFiles/opencpn.dir/src/cat_settings.cpp.o CMakeFiles/opencpn.dir/src/certificates.cpp.o CMakeFiles/opencpn.dir/src/chartdb.cpp.o CMakeFiles/opencpn.dir/src/chartdbs.cpp.o CMakeFiles/opencpn.dir/src/chartimg.cpp.o CMakeFiles/opencpn.dir/src/chcanv.cpp.o CMakeFiles/opencpn.dir/src/ChInfoWin.cpp.o CMakeFiles/opencpn.dir/src/color_handler.cpp.o CMakeFiles/opencpn.dir/src/compass.cpp.o CMakeFiles/opencpn.dir/src/concanv.cpp.o CMakeFiles/opencpn.dir/src/ConfigMgr.cpp.o CMakeFiles/opencpn.dir/src/conn_params_panel.cpp.o CMakeFiles/opencpn.dir/src/connections_dialog.cpp.o CMakeFiles/opencpn.dir/src/DetailSlider.cpp.o CMakeFiles/opencpn.dir/src/download_mgr.cpp.o CMakeFiles/opencpn.dir/src/FlexHash.cpp.o CMakeFiles/opencpn.dir/src/FontDesc.cpp.o CMakeFiles/opencpn.dir/src/FontMgr.cpp.o CMakeFiles/opencpn.dir/src/GoToPositionDialog.cpp.o CMakeFiles/opencpn.dir/src/gshhs.cpp.o CMakeFiles/opencpn.dir/src/gui_lib.cpp.o CMakeFiles/opencpn.dir/src/IDX_entry.cpp.o CMakeFiles/opencpn.dir/src/iENCToolbar.cpp.o CMakeFiles/opencpn.dir/src/kml.cpp.o CMakeFiles/opencpn.dir/src/Layer.cpp.o CMakeFiles/opencpn.dir/src/LinkPropDlg.cpp.o CMakeFiles/opencpn.dir/src/load_errors_dlg.cpp.o CMakeFiles/opencpn.dir/src/logger.cpp.o CMakeFiles/opencpn.dir/src/MarkInfo.cpp.o CMakeFiles/opencpn.dir/src/mbtiles.cpp.o CMakeFiles/opencpn.dir/src/mDNS_query.cpp.o CMakeFiles/opencpn.dir/src/mDNS_service.cpp.o CMakeFiles/opencpn.dir/src/MUIBar.cpp.o CMakeFiles/opencpn.dir/src/multiplexer.cpp.o CMakeFiles/opencpn.dir/src/navutil.cpp.o CMakeFiles/opencpn.dir/src/NMEALogWindow.cpp.o CMakeFiles/opencpn.dir/src/ocpCursor.cpp.o CMakeFiles/opencpn.dir/src/ocpn_app.cpp.o CMakeFiles/opencpn.dir/src/OCPN_AUIManager.cpp.o CMakeFiles/opencpn.dir/src/ocpndc.cpp.o CMakeFiles/opencpn.dir/src/ocpn_frame.cpp.o CMakeFiles/opencpn.dir/src/OCPNListCtrl.cpp.o CMakeFiles/opencpn.dir/src/ocpn_pixel.cpp.o CMakeFiles/opencpn.dir/src/OCPNPlatform.cpp.o CMakeFiles/opencpn.dir/src/ocpn_plugin.cpp.o CMakeFiles/opencpn.dir/src/ocpn_print.cpp.o CMakeFiles/opencpn.dir/src/OCPNRegion.cpp.o CMakeFiles/opencpn.dir/src/options.cpp.o CMakeFiles/opencpn.dir/src/peer_client.cpp.o CMakeFiles/opencpn.dir/src/piano.cpp.o CMakeFiles/opencpn.dir/src/pluginmanager.cpp.o CMakeFiles/opencpn.dir/src/plugin_paths.cpp.o CMakeFiles/opencpn.dir/src/printtable.cpp.o CMakeFiles/opencpn.dir/src/priority_gui.cpp.o CMakeFiles/opencpn.dir/src/Quilt.cpp.o CMakeFiles/opencpn.dir/src/REST_server_gui.cpp.o CMakeFiles/opencpn.dir/src/RolloverWin.cpp.o CMakeFiles/opencpn.dir/src/routemanagerdialog.cpp.o CMakeFiles/opencpn.dir/src/routeman_gui.cpp.o CMakeFiles/opencpn.dir/src/route_gui.cpp.o CMakeFiles/opencpn.dir/src/route_point_gui.cpp.o CMakeFiles/opencpn.dir/src/routeprintout.cpp.o CMakeFiles/opencpn.dir/src/RoutePropDlg.cpp.o CMakeFiles/opencpn.dir/src/RoutePropDlgImpl.cpp.o CMakeFiles/opencpn.dir/src/S57QueryDialog.cpp.o CMakeFiles/opencpn.dir/src/safe_mode.cpp.o CMakeFiles/opencpn.dir/src/SendToGpsDlg.cpp.o CMakeFiles/opencpn.dir/src/SendToPeerDlg.cpp.o CMakeFiles/opencpn.dir/src/Station_Data.cpp.o CMakeFiles/opencpn.dir/src/styles.cpp.o CMakeFiles/opencpn.dir/src/svg_utils.cpp.o CMakeFiles/opencpn.dir/src/TCDataFactory.cpp.o CMakeFiles/opencpn.dir/src/TCDataSource.cpp.o CMakeFiles/opencpn.dir/src/TCDS_Ascii_Harmonic.cpp.o CMakeFiles/opencpn.dir/src/TCDS_Binary_Harmonic.cpp.o CMakeFiles/opencpn.dir/src/tcmgr.cpp.o CMakeFiles/opencpn.dir/src/TCWin.cpp.o CMakeFiles/opencpn.dir/src/thumbwin.cpp.o CMakeFiles/opencpn.dir/src/toolbar.cpp.o CMakeFiles/opencpn.dir/src/track_gui.cpp.o CMakeFiles/opencpn.dir/src/trackprintout.cpp.o CMakeFiles/opencpn.dir/src/TrackPropDlg.cpp.o CMakeFiles/opencpn.dir/src/TTYScroll.cpp.o CMakeFiles/opencpn.dir/src/TTYWindow.cpp.o CMakeFiles/opencpn.dir/src/undo.cpp.o CMakeFiles/opencpn.dir/src/update_mgr.cpp.o CMakeFiles/opencpn.dir/src/waypointman_gui.cpp.o CMakeFiles/opencpn.dir/src/viewport.cpp.o CMakeFiles/opencpn.dir/src/shaders.cpp.o CMakeFiles/opencpn.dir/src/linux_devices.cpp.o CMakeFiles/opencpn.dir/src/udev_rule_mgr.cpp.o CMakeFiles/opencpn.dir/src/ais_bitstring.cpp.o CMakeFiles/opencpn.dir/src/ais_decoder.cpp.o CMakeFiles/opencpn.dir/src/ais_target_data.cpp.o CMakeFiles/opencpn.dir/src/base_platform.cpp.o CMakeFiles/opencpn.dir/src/catalog_handler.cpp.o CMakeFiles/opencpn.dir/src/catalog_parser.cpp.o CMakeFiles/opencpn.dir/src/chartdata_input_stream.cpp.o CMakeFiles/opencpn.dir/src/comm_ais.cpp.o CMakeFiles/opencpn.dir/src/comm_navmsg.cpp.o CMakeFiles/opencpn.dir/src/comm_navmsg_bus.cpp.o CMakeFiles/opencpn.dir/src/comm_appmsg.cpp.o CMakeFiles/opencpn.dir/src/comm_appmsg_bus.cpp.o CMakeFiles/opencpn.dir/src/comm_bridge.cpp.o CMakeFiles/opencpn.dir/src/comm_decoder.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_file.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_factory.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_n0183.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_n0183_net.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_n0183_serial.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_n2k.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_n2k_serial.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_registry.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_signalk.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_signalk_net.cpp.o CMakeFiles/opencpn.dir/src/comm_n0183_output.cpp.o CMakeFiles/opencpn.dir/src/comm_util.cpp.o CMakeFiles/opencpn.dir/src/comm_vars.cpp.o CMakeFiles/opencpn.dir/src/comm_plugin_api.cpp.o CMakeFiles/opencpn.dir/src/conn_params.cpp.o CMakeFiles/opencpn.dir/src/config_vars.cpp.o CMakeFiles/opencpn.dir/src/cutil.cpp.o CMakeFiles/opencpn.dir/src/downloader.cpp.o CMakeFiles/opencpn.dir/src/garmin_protocol_mgr.cpp.o CMakeFiles/opencpn.dir/src/geodesic.cpp.o CMakeFiles/opencpn.dir/src/georef.cpp.o CMakeFiles/opencpn.dir/src/hyperlink.cpp.o CMakeFiles/opencpn.dir/src/nav_object_database.cpp.o CMakeFiles/opencpn.dir/src/navutil_base.cpp.o CMakeFiles/opencpn.dir/src/ocpn_utils.cpp.o CMakeFiles/opencpn.dir/src/own_ship.cpp.o CMakeFiles/opencpn.dir/src/plugin_blacklist.cpp.o CMakeFiles/opencpn.dir/src/plugin_cache.cpp.o CMakeFiles/opencpn.dir/src/plugin_handler.cpp.o CMakeFiles/opencpn.dir/src/plugin_loader.cpp.o CMakeFiles/opencpn.dir/src/position_parser.cpp.o CMakeFiles/opencpn.dir/src/REST_server.cpp.o CMakeFiles/opencpn.dir/src/route.cpp.o CMakeFiles/opencpn.dir/src/route_point.cpp.o CMakeFiles/opencpn.dir/src/routeman.cpp.o CMakeFiles/opencpn.dir/src/select.cpp.o CMakeFiles/opencpn.dir/src/select_item.cpp.o CMakeFiles/opencpn.dir/src/semantic_vers.cpp.o CMakeFiles/opencpn.dir/src/ser_ports.cpp.o CMakeFiles/opencpn.dir/src/track.cpp.o CMakeFiles/opencpn.dir/src/comm_drv_n2k_socketcan.cpp.o CMakeFiles/opencpn.dir/src/garmin_wrapper.cpp.o CMakeFiles/opencpn.dir/src/glTextureDescriptor.cpp.o CMakeFiles/opencpn.dir/src/glTexCache.cpp.o CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o CMakeFiles/opencpn.dir/src/glTextureManager.cpp.o -o opencpn  -lGLEW  libs/observable/lib_OBSERVABLE.a  libs/ssl_sha1/libSHA1.a  libs/mongoose/libMONGOOSE.a  libs/mDNS/libMDNS.a  libs/SQLiteCpp/libSQLITE_CPP.a  -lbz2  -lz  -lSM  -lICE  -lX11  -lXext  libs/wxcurl/libWXCURL.a  -lcurl  [... cut ...]
   35 |         struct {
      | 
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/x86_64-pc-linux-gnu-g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Downstream report: https://bugs.gentoo.org/927484
Full build log: [build.log](https://github.com/OpenCPN/OpenCPN/files/14861681/build.log)

leamas, 2024-04-08:
So the problem found here is basically that `garmin_usb_packet` is defined on both _libs/garmin/jeeps/garminusb.h_ and _model/include/model/garmin_protocol_mgr.h_. 

Of course this violates all sorts of common sense including the "C++ One Definition Rule". We need to fix this. However, I fear this is not the only one.

leamas, 2024-04-08:
@eli-schwartz : You might want to build using attached patch.  Will try to make it into  master later.
[fix-duplicate-defs.patch.gz](https://github.com/OpenCPN/OpenCPN/files/14911036/fix-duplicate-defs.patch.gz)

eli-schwartz, 2024-04-08:
Thanks. The patch doesn't cleanly backport to the stable release so I attempted building from git master instead. I have a different error now:

```
FAILED: cli/opencpn-cmd 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -pthread -D__WXGTK3__ -Wl,-O1 -Wl,--as-needed -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 model/CMakeFiles/_model_src.dir/src/ais_bitstring.cpp.o model/CMakeFiles/_model_src.dir/src/ais_decoder.cpp.o model/CMakeFiles/_model_src.dir/src/ais_state_vars.cpp.o model/CMakeFiles/_model_src.dir/src/ais_target_data.cpp.o model/CMakeFiles/_model_src.dir/src/base_platform.cpp.o model/CMakeFiles/_model_src.dir/src/catalog_handler.cpp.o model/CMakeFiles/_model_src.dir/src/catalog_parser.cpp.o model/CMakeFiles/_model_src.dir/src/certificates.cpp.o model/CMakeFiles/_model_src.dir/src/chartdata_input_stream.cpp.o model/CMakeFiles/_model_src.dir/src/cli_platform.cpp.o model/CMakeFiles/_model_src.dir/src/cmdline.cpp.o model/CMakeFiles/_model_src.dir/src/comm_ais.cpp.o model/CMakeFiles/_model_src.dir/src/comm_appmsg_bus.cpp.o model/CMakeFiles/_model_src.dir/src/comm_appmsg.cpp.o model/CMakeFiles/_model_src.dir/src/comm_bridge.cpp.o model/CMakeFiles/_model_src.dir/src/comm_can_util.cpp.o model/CMakeFiles/_model_src.dir/src/comm_decoder.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_factory.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_file.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n0183.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n0183_net.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n0183_serial.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n2k.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n2k_net.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n2k_serial.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_registry.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_signalk.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_signalk_net.cpp.o model/CMakeFiles/_model_src.dir/src/comm_n0183_output.cpp.o model/CMakeFiles/_model_src.dir/src/comm_navmsg_bus.cpp.o model/CMakeFiles/_model_src.dir/src/comm_navmsg.cpp.o model/CMakeFiles/_model_src.dir/src/comm_util.cpp.o model/CMakeFiles/_model_src.dir/src/comm_vars.cpp.o model/CMakeFiles/_model_src.dir/src/config_vars.cpp.o model/CMakeFiles/_model_src.dir/src/conn_params.cpp.o model/CMakeFiles/_model_src.dir/src/cutil.cpp.o model/CMakeFiles/_model_src.dir/src/downloader.cpp.o model/CMakeFiles/_model_src.dir/src/garmin_protocol_mgr.cpp.o model/CMakeFiles/_model_src.dir/src/geodesic.cpp.o model/CMakeFiles/_model_src.dir/src/georef.cpp.o model/CMakeFiles/_model_src.dir/src/gui.cpp.o model/CMakeFiles/_model_src.dir/src/hyperlink.cpp.o model/CMakeFiles/_model_src.dir/src/instance_handler.cpp.o model/CMakeFiles/_model_src.dir/src/ipc_api.cpp.o model/CMakeFiles/_model_src.dir/src/local_api.cpp.o model/CMakeFiles/_model_src.dir/src/logger.cpp.o model/CMakeFiles/_model_src.dir/src/mDNS_query.cpp.o model/CMakeFiles/_model_src.dir/src/mDNS_service.cpp.o model/CMakeFiles/_model_src.dir/src/multiplexer.cpp.o model/CMakeFiles/_model_src.dir/src/nav_object_database.cpp.o model/CMakeFiles/_model_src.dir/src/navutil_base.cpp.o model/CMakeFiles/_model_src.dir/src/ocpn_plugin.cpp.o model/CMakeFiles/_model_src.dir/src/ocpn_utils.cpp.o model/CMakeFiles/_model_src.dir/src/own_ship.cpp.o model/CMakeFiles/_model_src.dir/src/peer_client.cpp.o model/CMakeFiles/_model_src.dir/src/pincode.cpp.o model/CMakeFiles/_model_src.dir/src/plugin_api.cpp.o model/CMakeFiles/_model_src.dir/src/plugin_blacklist.cpp.o model/CMakeFiles/_model_src.dir/src/plugin_cache.cpp.o model/CMakeFiles/_model_src.dir/src/plugin_handler.cpp.o model/CMakeFiles/_model_src.dir/src/plugin_loader.cpp.o model/CMakeFiles/_model_src.dir/src/plugin_paths.cpp.o model/CMakeFiles/_model_src.dir/src/position_parser.cpp.o model/CMakeFiles/_model_src.dir/src/rest_server.cpp.o model/CMakeFiles/_model_src.dir/src/route.cpp.o model/CMakeFiles/_model_src.dir/src/routeman.cpp.o model/CMakeFiles/_model_src.dir/src/route_point.cpp.o model/CMakeFiles/_model_src.dir/src/safe_mode.cpp.o model/CMakeFiles/_model_src.dir/src/select.cpp.o model/CMakeFiles/_model_src.dir/src/select_item.cpp.o model/CMakeFiles/_model_src.dir/src/semantic_vers.cpp.o model/CMakeFiles/_model_src.dir/src/ser_ports.cpp.o model/CMakeFiles/_model_src.dir/src/track.cpp.o model/CMakeFiles/_model_src.dir/src/usb_watch_factory.cpp.o model/CMakeFiles/_model_src.dir/src/wx_instance_chk.cpp.o model/CMakeFiles/_model_src.dir/src/std_instance_chk.cpp.o model/CMakeFiles/_model_src.dir/src/ipc_factories.cpp.o model/CMakeFiles/_model_src.dir/src/comm_drv_n2k_socketcan.cpp.o model/CMakeFiles/_model_src.dir/src/dbus_client.cpp.o model/CMakeFiles/_model_src.dir/src/dbus_server.cpp.o model/CMakeFiles/_model_src.dir/src/linux_usb_watch.cpp.o model/CMakeFiles/_model_src.dir/src/garmin_wrapper.cpp.o cli/CMakeFiles/opencpn-cmd.dir/console.cpp.o cli/CMakeFiles/opencpn-cmd.dir/api_shim.cpp.o -o cli/opencpn-cmd  -Wl,-rpath,/var/tmp/portage/sci-geosciences/opencpn-5.8.4-r1/work/opencpn-5.8.4_build/libs/IXWebSocket:  /usr/lib64/libGLEW.so  -rdynamic  libs/wxcurl/libWXCURL.a  -L/usr/lib64  -pthread  -lwx_gtk3u_gl-3.2-gtk3  -lwx_baseu_net-3.2-gtk3  -lwx_baseu_xml-3.2-gtk3  -lwx_gtk3u_html-3.2-gtk3  -l [... cut ...]
/var/tmp/portage/sci-geosciences/opencpn-5.8.4-r1/work/opencpn-5.8.4/model/src/comm_drv_signalk_net.cpp:443:(.text+0x9ad1): undefined reference to `ix::initNetSystem()'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: /var/tmp/portage/sci-geosciences/opencpn-5.8.4-r1/temp/cchCCMTl.ltrans70.ltrans.o: in function `MyFrame::OnCloseWindow(wxCloseEvent&)':
/var/tmp/portage/sci-geosciences/opencpn-5.8.4-r1/work/opencpn-5.8.4/model/src/comm_drv_signalk_net.cpp:447:(.text+0x98c8): undefined reference to `ix::uninitNetSystem()'
collect2: error: ld returned 1 exit status
ninja: build stopped: cannot make progress due to previous errors.
```

leamas, 2024-04-08:
That looks like some kind of build problem on your side. Current master  + that patch builds cleanly for me, and the linkage errors seems unrelated. In particular, the WebSocker library is bundled so it should certainly not be missing.

eli-schwartz, 2024-04-08:
As you can see, it's attempting to link to libs/IXWebSocket/libixwebsocket.so.11.3.2, a shared library. It's not missing.

```
$ nm -D --demangle /var/tmp/portage/sci-geosciences/opencpn-5.8.4-r1/work/opencpn-5.8.4_build/libs/IXWebSocket/libixwebsocket.so.11.3.2 
                 U __cxa_atexit@GLIBC_2.2.5
                 w __cxa_finalize@GLIBC_2.2.5
0000000000000000 A __gentoo_check_ldflags__
                 w __gmon_start__
                 U __gxx_personality_v0@CXXABI_1.3
                 w _ITM_deregisterTMCloneTable
                 w _ITM_registerTMCloneTable
                 U memcpy@GLIBC_2.14
                 U __stack_chk_fail@GLIBC_2.4
                 U strlen@GLIBC_2.2.5
                 U operator delete(void*)@GLIBCXX_3.4
                 U std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)@GLIBCXX_3.4.21
                 U std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()@GLIBCXX_3.4.21
                 U std::__throw_logic_error(char const*)@GLIBCXX_3.4
                 U std::ios_base_library_init()@GLIBCXX_3.4.32
```
This is not a build problem on my side above and beyond the fact that LTO tried to optimize the library a bit too much :D

eli-schwartz, 2024-04-08:
It is a linux distro environment. BUILD_SHARED_LIBS=ON is automatically set by the underlying infrastructure for CMake based builds, since for software that provides public libraries, shared is preferred over static.

Unfortunately, the cmake build system is absolutely HORRIBLE about this. :(
- GNU autotools has `--enable-static --enable-shared`
- meson has `-Ddefault_library={static|shared|both}`

and you can specify inside a project, that you specifically need an uninstalled "convenience library" that is always static, or that you don't support static builds at all. CMake just sort of makes you guess.

For meson it's so much easier, the distro tooling just specifies `-Ddefault_library=shared` and applications that are building internal static libraries define them with `static_library('ixwebsocket', 'foo.c')` instead of `library('ixwebsocket', 'foo.c')` to avoid default-promoting to shared libraries. There is no working heuristic for cmake.

What really confuses me is I found 22 static archives in the build directory for OpenCPN, and 3 shared libraries. libixwebsocket.so even contains a versioned soname. It simply did not cross my mind in the slightest that it wasn't *meant* to be a shared library.

leamas, 2024-04-08:
@eli-schwartz : You can find a number of Debian/Ubuntu builds  [here](https://app.circleci.com/pipelines/github/leamas/OpenCPN/4936/workflows/5e7af3cf-d35a-46c1-951c-384d27a68fc1)

On my Fedora box, with build deps in place, I just do 

    $ mkdir build; cd build
    $ cmake ..
    $ make -sj12

Which builds everything. In particular, it builds static libraries from the libraries in build/. And I would say this is what you want also from a system perspective -- these libraries are patched and should only be visible for opencpn.

eli-schwartz, 2024-04-08:
Well, yes, the problem is the `..` part. ;)

I must say, having bundled libs feels like the root of the problem here. In general this is not-good news. For example, what is wrong with simply using the externally available ixwebsocket if detected on the system?

> these libraries are patched

are they forks, containing novel changes not available upstream? Are the changes documented somewhere? How do you sync with upstream fixes?

For ixwebsocket in particular, it is only included in git master and not in a stable release. If it had made it into a stable release, it would surely have been noticed, because in the current state of affairs, OpenCPN tries to install:
- /usr/include/ixwebsocket
- /usr/lib64/libixwebsocket.a
- /usr/lib64/pkgconfig/ixwebsocket.pc
- /usr/lib64/cmake/ixwebsocket/

So:

> these libraries are patched and should only be visible for opencpn.

No, no, no, it is absolutely not only visible for OpenCPN, and until this is resolved so that at *minimum* it does not install files that conflict with an existing systemwide ixwebsocket, the next version of OpenCPN is simply unpackageable by linux distros.

nohal, 2024-04-08:
> Well, yes, the problem is the `..` part. ;)
>
> I must say, having bundled libs feels like the root of the problem here. In general this is not-good news. For example, what is wrong with simply using [...]

They do not contain any novel changes deserving a fork, usually only fixes to make them work on all the platforms we need to support.
We do sync with upstream fixes as required when our problems are addressed there.
Generally we try to use upstream directly on Linux when it is available in major distros and we sure welcome pull requests to do so for the libraries where we do not know about it.

> For ixwebsocket in particular, it is only included in git master and not in a stable release. If it had made it into a stable release, it would surely have been noticed, because in the current state [...]

Installing artifacts of any of the bundled libraries sure is a bug, not intended.

eli-schwartz, 2024-04-09:
> Generally we try to use upstream directly on Linux when it is available in major distros and we sure welcome pull requests to do so for the libraries where we do not know about it.

IMHO it can't hurt to pre-emptively offer the option for every dependency where it's possible. :)

I opened a gentoo tracking ticket for this to be reviewed so that pull requests could be submitted.

I also manually passed BUILD_SHARED_LIBS=OFF and rebuilt with LTO and the linked PR. It succeeded, which indicates that as far as this ticket's original purpose is concerned:

Yes, the linked PR fixes LTO for me.

eli-schwartz, 2024-04-09:
> No, no, no, it is absolutely not only visible for OpenCPN, and until this is resolved so that at _minimum_ it does not install files that conflict with an existing systemwide ixwebsocket, the next version of OpenCPN is simply unpackageable by linux distros.

And this is solved by #3797, thanks.

nohal, 2024-04-09:
Please reopen if the build system is still not happy...

PR #3795 (2024-04-09): Remove duplicated definitions (#3782)
Remove some duplicated defintions revealed using `-flto=4`.

EDIT: Closes: #3782
closed 2024-04-09 by nohal