OpenCPN/OpenCPN #3785: The docked dashboard width setting is not saved

bc38, 2024-04-04:
Configured
[image]

After turning off and on the plugin on the dashboard
[image]

bdbcat, 2024-04-22:
Ready for close.

bc38, 2024-04-22:
the problem is not completely solved!
Enable the dashboard, attach it, adjust the width.
Disable the plugin and close CPN
Launch CPN and enable the dashboard plugin.
The dashboard width has been reset to default.

bc38, 2024-04-23:
Works great!
Thank you

bc38, 2024-09-29:
Dashboard does not save panel width again Reproduce: enable dashboard, adjust width, hide dashboard, reboot CPN.
After rebooting CPN and enabling dashboard, panel width will be default.

bdbcat, 2024-10-01:
Should be corrected in github master now.

closed 2024-10-02 by bc38