OpenCPN/OpenCPN #3786: Add TinyXML2 to opencpn-libs

Rasbats, 2024-04-05:
TinyXML has not been updated for about 10 years and is no longer maintained.

Suggest adding TinyXML2 to the libraries available from opencpn-libs.

TinyXML2 simplifies the construction of xml files.

nohal, 2024-04-05:
Isn't it better to use `pugixml`, which is already available, faster and used by the core OpenCPN?

Rasbats, 2024-04-05:
I have been updating sar_pi, which used TinyXML. I moved to TinyXML2. 
It is maintained. Last commit about 4 months ago. 
There are quite a few plugins using TinyXML. Perhaps putting pugixml in opencpn-libs would encourage developers to move in that direction.

nohal, 2024-04-05:
`pugixml` is in `openpcn-libs`, that's why we are suggesting it, https://github.com/OpenCPN/opencpn-libs/tree/main/pugixml

Rasbats, 2024-04-05:
Sorry, missed that.

closed 2024-04-05 by Rasbats

rgleason, 2024-04-05:
Almost all TP plugins use tinyxml

Rasbats, 2024-04-09:
And pugixml in opencpn-libs is 8 years old, v1.8. Current version is v1.14.

leamas, 2024-04-09:
Ah... time to update!