OpenCPN/OpenCPN #3788: Issue for improvement: ENC S63 chart performance on rasperry pi 4

placebojim, 2024-04-06:
There is an issue with ENC S63 charts, they are extremely slow compared to raster or ENC S57 charts. Performance numbers on a raspberry pi 4 for a slow pan with a mouse are as follows:

With raster charts for NZ performance is good - OpenGL frame time ~20ms, opencpn 40% CPU

With s57 chart set load from chart downloader tab for US NW performance is good OpenGL frame time ~30ms, opencpn 50% CPU

With a single ENC chart loaded using the s63 plugin performance is nearly unuseable. OpenGL frame time ~700ms, opencpn at 100% CPU. Very laggy and some seconds behind the mouse panning movement. The only practical panning method is to use keyboard keys as this keeps the window changes down to a level that the system can cope with. Reducing chart detail to bare minimum doesn't help.

This is tested on opencpn 5.8.4 and s63 plugin version 1.17.15. Performance here is slightly worse with Software OpenGL ticked on the raster and S57 case but the same with the S63 case. I have also tried changing from dtoverlay=vc4-kms-v3d to dtoverlay=vc4-fkms-v3d but no change. This doesn't seem like a graphics issue but perhaps something to do with the decryption?

I am running the latest open plotter image Debian bullseye 64 bit
> uname -a
Linux sliverpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

leamas, 2024-05-06:
This is a good bug to work with in next cycle. Reopening and tagging for 5.12

Thanks for  reporting!

placebojim, 2024-05-07:
This is a duplicate of #3789. I only closed this to open it as a suggestion not a bug.

leamas, 2024-05-07:
Ah, ok, m bad. Closing

closed 2024-05-07 by leamas