OpenCPN/OpenCPN #3789: Improve ENC S63 chart performance on rasperry pi 4

placebojim, 2024-04-06:
There is an issue with ENC S63 charts, they are extremely slow compared to raster or ENC S57 charts. Performance numbers on a raspberry pi 4 for a slow pan with a mouse are as follows:

With raster charts for NZ performance is good - OpenGL frame time ~20ms, opencpn 40% CPU

With s57 chart set load from chart downloader tab for US NW performance is good OpenGL frame time ~30ms, opencpn 50% CPU

With a single ENC chart loaded using the s63 plugin performance is nearly unuseable. OpenGL frame time ~700ms, opencpn at 100% CPU. Very laggy and some seconds behind the mouse panning movement. The only practical panning method is to use keyboard keys as this keeps the window changes down to a level that the system can cope with. Reducing chart detail to bare minimum doesn't help.

This is tested on opencpn 5.8.4 and s63 plugin version 1.17.15. Performance here is slightly worse with Software OpenGL ticked on the raster and S57 case but the same with the S63 case. I have also tried changing from dtoverlay=vc4-kms-v3d to dtoverlay=vc4-fkms-v3d but no change. This doesn't seem like a graphics issue but perhaps something to do with the decryption?

I am running the latest open plotter image Debian bullseye 64 bit
> uname -a
Linux sliverpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

LordLaingholm, 2024-05-06:
I have the same issue with S63 charts, I use NZ charts and there's always a 1-2 sec delay panning with the mouse and it's slightly slower on a touchscreen, I just put up with it but I'm keen to see if you get any traction on this as it is annoying.

btw my config is pi 4 Opencpn 5.8.4-1 on Linux openplotter

placebojim, 2024-05-07:
Duplicate of #3788

placebojim, 2024-05-07:
@leamas would be great if this makes it into OpenCPN 5.12.0 :)

leamas, 2024-05-08:
Indeed it would.

JeremyDWilliams, 2024-06-30:
As it's a main function of the software to use a chart dynamically, it would seem that this should be a priority.

DigNZ, 2026-09-29:
Chipping in with a new Windows install  5.14.2 and noticing the same thing - S63 (NZ charts from LINZ) pan and zoom much slower than other chart types. No CPU or GPU spike measured and consistent low memory footprint. One thing I did notice was that it pans smoother with the arrow keys than the mouse. Is it possible that mouse events are triggering processes that cause a wait. Keypress and repeat is slow enough that the waits don't queue up?