OpenCPN/OpenCPN #3790: Connection panel cannot be unselected

transmitterdan, 2024-04-07:
**Describe the bug**
In "Connections" dialog once a panel is selected it cannot be unselected. This is important because when a panel is selected it disables the "Add Connection" button.  Thus, once a panel is clicked it's impossible to add a new connection.

Why is important to disable "Add Connection" button.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options->Connections 
2. Click on an existing connection panel
3. "Add Connection" is greyed with no way to make it active again.
[image]

**Expected behavior**
It should be possible to add a new connection.

It seems this line is what determines that "Add Connection" is disabled: https://github.com/OpenCPN/OpenCPN/blob/53aab42b83e26ae6680a5a797984ea315fc17c59/gui/src/connections_dialog.cpp#L379

Once "panel" bool is set there is no way to unset it.

transmitterdan, 2024-04-07:
Sometimes it works.  Closing until more research done.

closed 2024-04-07 by transmitterdan

TwoCanPlugIn, 2024-04-07:
It's an intuitive "toggle"

Click the connection again to deselect it, and the "Add Connection" button is enabled.

transmitterdan, 2024-04-07:
It should also work if you click somewhere else in the panel.  It's not as intuitive as it should be.

TwoCanPlugIn, 2024-04-07:
Bugger, my sarcasm is being lost in the ether.

nohal, 2024-04-07:
> Bugger, my sarcasm is being lost in the ether.

Yes, your sarcastic rants are useless, good that even you finally realized it.

rgleason, 2024-04-07:
It cant be too bad. I figured this out. Probably could be fixed, for those that get puzzled by it.