OpenCPN/OpenCPN #3792: The (SE)vector palette can create a crash. Do we need to update these for O 5.10?

Hakansv, 2024-04-07:
I won't file it a bug since the actual (SE) vector palette is involved and may need to be edit to fit O 5.10?

On Windows using o-charts and the vector palette downloaded [here](https://www.opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:charts:vector_palette#swedish_vector_palette) creates a O crash when zooming to some detailed charts.
FYI is the inbound "chartsymbols" and "rastersymbols-day" not creating any crash for the same location.
When zooming into a detailed chart:  e.g oeuSENC-DK/OC-45-KBLAA5.oesu
I get this crash: (o-charts not loaded to debug so no details from there) Pls advice if I should build and include o-chart in the debug?

[image]

Call stack:
```
 	o-charts_pi.dll!6d800486()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for o-charts_pi.dll]	
 	[External Code]	
>	opencpn.exe!ChartPlugInWrapper::RenderRegionViewOnGLNoText(const wxGLContext & glc, const ViewPort & VPoint, const OCPNRegion & RectRegion, const LLRegion & Region) Line 5550	C++
 	opencpn.exe!glChartCanvas::RenderQuiltViewGL(ViewPort & vp, const OCPNRegion & rect_region) Line 3286	C++
 	opencpn.exe!glChartCanvas::RenderCharts(ocpnDC & dc, const OCPNRegion & rect_region) Line 3549	C++
 	opencpn.exe!glChartCanvas::Render() Line 4262	C++
 	opencpn.exe!glChartCanvas::OnPaint(wxPaintEvent & event) Line 1440	C++
 	[External Code]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 738	C++
 	[External Code]	
```

bdbcat, 2024-04-07:
Do you agree that this is an o-charts bug only?  Never seen on inbound ENC?

Hakansv, 2024-04-07:
Yes, agree.
btw: I think I've been able to edit chartsymbols..xml for SE vector palette to fit in 5.9. wo crash.
What I can see is the present rastersymbols-day.png for SE palette still usable.
I'll check more and come back....

Hakansv, 2024-04-08:
I've working vector palette files for O 5.10 (and o-charts 2.0.0.66)
@rgleason Would you mind copying the file below to the [manual](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:charts:vector_palette#swedish_vector_palette)
I've prepared for it here:  _".......for OCPN v 5.10 and never can be downloaded here:..."_
Thanks
[S57data_SE_palette_O5.10.zip](https://github.com/OpenCPN/OpenCPN/files/14906940/S57data_SE_palette_O5.10.zip)

Hakansv, 2024-04-09:
The vector palette file is now in the [manual](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:charts:vector_palette)

closed 2024-04-09 by Hakansv