OpenCPN/OpenCPN #3800: Redundant or inconsistent UI - Unis (distance)

TwoCanPlugIn, 2024-04-09:
The user selects distance units under Options->Display.>Units.

If the user then goes to Options->Ships->Own Ship, the show range rings setting allows the user choose units.

Similar for Options->Ships.>Routes/Points, waypoint range rings.

BTW, the "Show Direction to Active Waypoint" has two settings, Style & Colour. Both default to "default". What is the default Style & Line Colour ?

See below for each.

[image]

[image]

bdbcat, 2024-04-22:
NotABug..
Converting to "Enhancement"

TwoCanPlugIn, 2024-04-22:
You say potato, I say potatoe!

OK, not what I would call a Severity "A"bug.