OpenCPN/OpenCPN #3801: NMEA 0183 Connection, Talker ID is not persisted.

TwoCanPlugIn, 2024-04-09:
If a NMEA 0183 Connection is set as "Output on this port", if I change the Talker ID, it is not persisted.

Is this also in some way related to the outbound filter selections where there are ECAPB, ECRMB etc.
[image]

Hakansv, 2024-04-17:
What I can see, correct me if I'm wrong, is that the value in control "Talker ID" is only used for a serial output when "Format uploads for Furuno GP3X" is enabled. 
If so, the "Talker ID (Blank.... " would be hidden when "Furuno    " is not enabled?
In any case I can not see it's used to change Talker ID "EC" for sentences created by routeman for an AP, i.e. an activated route.

TwoCanPlugIn, 2024-04-17:
Another case  of discombobulated UI !

Without reading a manual, where it's probably not documented anyway, a user wouldn't understand the intent of the UI. If it is the way that you suggest it is, there is no relationship in the dialog between the Furuno checkbox and the Talker Id text box. Given the current UI layout It certainly doesn't behave the way that I would expect it to.

Hakansv, 2024-04-17:
I was trying to read the code. Not sure if I have succeeded with that. 
Our goals with issues would be to report strange behavior and possible suggest a solution.

bdbcat, 2024-04-18:
Looks like a clear bug, possibly introduced with new comm stack model.
But a bug, anyway.

TwoCanPlugIn, 2024-04-18:
In addition to that, what really doesn't make sense is that if the port is selected for output and the Output Filtering radio button is set, but the filter list is empty, everything gets sent.

I really don't understand the difference between "Transmit Sentences"or ""Drop Sentences"

Perhaps my view of logic differs from everyone else's ? To me what would make more sense is the attached mockup. (the filter list & browse button would only enabled if the checkbox is checked)

[image]

Hakansv, 2024-04-18:
@TwoCanPlugIn 
The logic is clear, at least to me:
If "Transmit sentences" is enabled and the filter field is empty than no filter is applied - send all i.e. no filtering.

The difference between "Transmit Sentences" and "Drop Sentences" is that the first transmit only those entered in filter field. The latter send all available except those entered in the filter field.  Again, if nothing is entered no filter is applied.

Hakansv, 2024-04-18:
@bdbcat 
The main problem why the  "Talker ID (Blank.... "  [EC] is not functional may be this:
If I open Edit connection and change from "EC" to something else, "HS" and click OK and then again open the Edit connection the "EC" is there again. So any changes are overwritten when I click "OK"

wholybee, 2024-04-20:
Maybe if there are no filters applied, instead of a blank wxTextCtrl, there could be a message "No Filter Applied"?

Hakansv, 2024-04-20:
(Viking? I suppose we're on about the same absolute parallel?)
The issue here would be the not functional change of Talker ID "EC"
I can't find anything in the code using this Connection parameter. Well one use is to immediately reset it to default when initializing the options window? But that's me, I can be wrong.
So either we make it working or take it out if not needed. No comments, except this issue, about the non working function is seen what I remember?
@bdbcat  Thoughts?

TwoCanPlugIn, 2024-04-20:
> Viking? I suppose we're on about the same absolute parallel?)

Actually I'm now in the same hemisphere, a bit lower and to the west of Greenwich!

The list box in the filters dialog is titled "Sentences", so in my mind that means NMEA 0183 sentences, (GLL, RMC, HDG etc.)  although I don't know whether that should include the talkerId. 

If there is an opportunity, perhaps one should revisit the use cases ? For example O receiving position from a multiplexer that has both a GPS receiver, AIS transceiver, Autopilot and VHF DSC radio.  Multiplexer receives position from both GPS receiver & AIS. Sends position from GPS receiver  to both VHF DSC and O and sends Autopilot commands to Autopilot. O definitely doesn't need to duplicate sending position fixes to VHF DSC.  Thoughts on use case ?

Also I'll reiterate why the need to have an inclusion list or an exclusion list. Surely one list makes sense ? Given sentences APB, RMB, XTE, HDM, VDM, VDO, GLL, GGA, GSV, RMC, XDR, WPL, an inclusion list of APB, RMB, XTE by definition excludes HDM, VDM, VDO, GLL, GGA, GSV, RMC, XDR and WPL. You are giving users two ways to do the one task. Two ways to make a mistake!

nohal, 2024-04-20:
There is no "need" for making the filter able to be defined both as inclusion and exclusion, an you are sure right that you can define a filter that lets only a single sentence pass also by defining 50 that are not allowed to pass (or vice versa). But I probably miss how that is supposed to be simpler and less error prone.

bdbcat, 2024-04-20:
I'm on this now, at least the OP item about talker ID.
Changes in filtering logic, if needed (IMHO, no) are a separate issue.
Tight focus, please.
Please stand by.

bdbcat, 2024-04-20:
First question:
Can we think of any reasonable use-case where we would NOT want the Talker-ID to be the same for all connections?
That is to say, can we design such that the Talker-ID is a global value?

Hakansv, 2024-04-20:
I can not see any need for a changed Talker-ID at all. So for me we can take it out. But I don't remember the reason for its implementation. 
If you decide to keep the function I think, yes we can design it such that Talker-ID is a global value.

wholybee, 2024-04-20:
I can't think of any reasonable use that requires a different talker id for different connections. I struggle to find a use at all.
Presumably, its so if there is conflicting data from multiple sources, the data from one source can be filter by talker ID?
I vote for making it global.

Hakansv, 2024-04-20:
@bdbcat 
Now the changed Talker-ID is reflected. Very good.
But for any reason is the multiplexer sorting out the three mentioned below.
Here I changed Talker to HS and included that in the filter but RED in NMEA DEbug.
ECXTE is though changed to HSXTE and passed through?
The black window to the left is Putty for what's tx-ed out.

Comment about changed code:
+    m_sentences.Add(g_TalkerIdText + wxString("RMB"));
+    m_sentences.Add(g_TalkerIdText + wxString("RMC"));
+    m_sentences.Add(g_TalkerIdText + wxString("APB"));

We should maybe handle also:
m_sentences.Add(g_TalkerIdText + wxString("XTE"));

I know XTE for any reason wasn't included before but routeman,cpp actually also produce ECXTE so we may include that as well?

[image]

Hakansv, 2024-04-20:
I wonder why XTE is working as the others although not included in the dotted list above?

bdbcat, 2024-04-20:
The actual TX code elsewhere uses the global Talker-ID, always has.
It is just the filter list that is missing xxXTE now.

Hakansv, 2024-04-20:
Ahaa. Didn't read code carefully enough.
Maybe add that as well for consistency? But not important.
Issue solved?

commit 6ba68bcc (2024-04-21): Add xxXTE NMEA0183 to output filter list. Closes #3801
closed 2024-04-21 by bdbcat