OpenCPN/OpenCPN #3803: User Manual:  Linux, Flatpak, RPI, MacOS  Installation & GNSS Instructions

rgleason, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
We need the manual updated for these OS = Linux, Flatpak, RPI, MacOS, 
I don't know the process, and contributors here are more authoritative 

**Describe the solution you'd like**
The installation pages are here
[Linux](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:linux)
[Flatpak](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:linux:flatpak)
[RPI](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:linux:rpi23)
[MacOS](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:macosx)
[Android](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:android)

The GNSS Setup pages are located here
[Linux](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps:linux)
[MacOS](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps:macos)
[RPI](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps:rpi)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

leamas, 2024-04-10:
First of all there is IMHO nothing like "RPi" or "Linux" in this context.  There is nothing special with RPi, it's just Linux.

OTOH, there is no such thing as "Linux". What we support is "Debian/Ubuntu" and "Flatpak".  This is true on all platforms, including the RPi. It also applies to the GNSS setup.

Furthermore, these are linked from [this page](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps) which is confusing, containing a lot of out of scope text probably best just removed.

We need a new GNSS structure where there is simple guides for the simple use case (USB connected GNSS device) for all platforms. This is basically about wizards except, perhaps, on Flatpak (#3799).

Separate from this there might be a **separate** chapter on more complicated setups. This is also probably less text than today.

Besides [this page](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:linux:plugins) should be removed.

leamas, 2024-04-10:
BTW: If the "GNSS Setup Pages" refers to the non-simple general usecase Windows is missing (OS-specific setups should of course not be on the main page).

rgleason, 2024-04-12:
I am going to close this as the discussion is continued in [Discussions User Manual Basic](https://github.com/OpenCPN/OpenCPN/discussions)   OK?

closed 2024-04-18 by rgleason