OpenCPN/OpenCPN #3805: Handle site palette configuration better

leamas, 2024-04-10:
**Is your feature request related to a problem?**
Some users for example Swedes like me wants to use a site-specific chart palette. This is described in the [user manual](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:charts:vector_palette). 

However, the method described means that this configuration is lost on a OpenCPN update.  Not nice.

**Describe the solution you'd like**
A user configured  palette should not be affected by an OpenCPN update.

**Describe alternatives you've considered**
The core change would be that OpenCPN  searched in some user and/or site path for the palette files before felling back to the palette which is part of the installation. Actual paths should not be a problem on any platform.

**Additional context**
Too late for 5.10

Hakansv, 2024-04-10:
> Anyway, if user uses a modified file it's her responsibility to maintain it. A change like this would just make sure that no user files are lost in an update.

That's why the manual not "Save the extracted files for possible future O update". 
Personally I have them in the /s57data folder but renamed to e.g. "chartsymbols_SE.xml" and they remains after updates.

leamas, 2024-04-10:
> Personally I have them in the /s57data folder but renamed to e.g. "chartsymbols_SE.xml" and they remains after updates.

You actually need three versions of the file to handle an update:

  1. The previous version OpenCPN version
  2.  Current OpenCPN version
  3. Your own, modified version.

With these you can set up a 3-way merge. It's basically about  applying the diff between 3) and 1) to 2).  It will always be manual.

If we store 3) in another location where it also actually is used this one is OK. 1) and 2) can be maintained automatically by OpenCPN installation scripts.

Don't forget that modifying files under .../S57  is more or less impossible on Linux. While "possible" on Windows and MacOS it's  still a bad practice. It's just that it's the only thing which works right now.