OpenCPN/OpenCPN #3807: SignalK Authentication

TwoCanPlugIn, 2024-04-10:
1. If the SignalK Server admin has disabled read only access, OpenCPN in its default configuration does not receive delta updates, as expected. However there is no visual indication other than the GPS satellite icon and nothing in the log that indicates that no data is received.
I expect O receives the SignalK Server announcement, which I presume it takes to mean everything is OK.

2. Authorization Token.
a. Do we really expect users to run signalk-generate-token ? 

b. Currently it does not seem to be persisted, nor retrieved from the configuration file. We can agree to disagree on the merits of storing a token that grants access to a remote device in clear text as discussed .  On Windows, in general you would use the Data Protection API (DPAPI) to securely store secrets. I'm not familiar with the corresponding API's on Mac or Linux. Mentioned towards the end of #3757

c. If an invalid or expired token is used, there is no indication of the authentication failure and no entry in the log. An invalid token will cause SignalK to respond with a HTTP 401 response upon the initial attempt to connect the web socket. 

At least the good news is that the authorization token works, so clearly O is setting the RequestHeader correctly.

nohal, 2024-04-10:
Perhaps https://github.com/mamba-org/libcred

TwoCanPlugIn, 2024-04-12:
Yes, looks like a good cross platform library. My Windows knowledge is a tad out of date, I'd forgotten about Credential Vault, which is probably better suited to this purpose. Can't comment on the other platforms.

However I don't understand the benefit of using an Authorization token.

nohal, 2024-04-12:
Benefit seems pretty clear to me:
- Without the token, impossible to connect to Signal K server with authentication enabled
- With the token, possible to connect to Signal K server with authentication enabled
Or did i misunderstood what you don'r understand?

rgleason, 2024-04-12:
I know it is in Windows 8, 10, 11   (don't remember about Win7)
[image]

bdbcat, 2024-04-22:
We have added signalK tracking in the NMEA Debug window.
Converting this issue to "Enhancement"

TwoCanPlugIn, 2024-04-22:
> We have added signalK tracking in the NMEA Debug window.

If the token has been incorrectly entered or the token has expired or been revoked, the NMEA Debug Window is empty. 

The user may not know why they are not receiving data, perhaps thinking it is a communications or SignalK error.

I don't think the logon error that SignalK returns (IIRC something like "statusCode":401) is displayed or logged by OpenCPN.

Peter-Petrik, 2025-03-10:
I created a user in SignalK with read-only permissions, then used the signalk-generate-token utility to generate a token.

Once I enter the token into OpenCPN, I can see the data from the connection. Until I restart OpenCPN, at which point the token in the field has been replaced with a "0" and no data is being received. Not sure where / how OpenCPN would store the value and retrieve it at next application launch, but in the current implementation it is not viable to expect the user to re-enter the token at each OpenCPN application launch.

Hakansv, 2025-03-11:
As described in a post at https://github.com/OpenCPN/OpenCPN/issues/4394#issuecomment-2713162350 I'd say the "Auth token" handling is working.

Peter-Petrik, 2025-03-11:
> As described in a post at [#4394 (comment)](https://github.com/OpenCPN/OpenCPN/issues/4394#issuecomment-2713162350) I'd say the "Auth token" handling is working.

The "Auth token" is working, until the OpenCPN app restarts. Then the token shows "0". So, technically the "Auth token" functionality is working, but in latest stable OpenCPN 5.10.2 there appears to be no mechanism to retrieve said token between app restarts.

Hakansv, 2025-03-11:
As of bdbcat's note above and my tests this bug is corrected. 
Of course the correction is made to Github master code in preparation for next O release.

Peter-Petrik, 2025-03-11:
> Of course the correction is made to Github master code in preparation for next O release.

Much appreciated @Hakansv. Is there an approximate timeline for the next anticipated release?

leamas, 2025-03-11:
This is an open source  project with the typical "we release when we are ready" promise.  That said, Q2 is an educated guess.

You might want to test the nightly builds at https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

Peter-Petrik, 2025-03-13:
Tested on MacOS 15.3.2 with build 5.11.0-universal+ab6f07d39 - the token entry persists between app restarts.