OpenCPN/OpenCPN #3808: O 5.9 master: Signalk not working. error: OpenSSL failed

Hakansv, 2024-04-11:
**Describe the bug**
SignalK connection don't work

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to Signalk.
2. No data received.
3. Check on browser that data is streaming.
4. Having O debug mode. See below

**Expected behavior**
Working SK connection.

**Desktop (please complete the following information if applicable):**
 - OS: Current master
 - Version 5.9

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Win 10

**Additional context**
```
11:02:33.966 MESSAGE comm_drv_factory.cpp:55 MakeCommDriver: Signal K:192.168.1.73:3000
11:02:33.966 MESSAGE comm_drv_signalk_net.cpp:335 Opening Signal K WebSocket client: Signal K:192.168.1.73:3000
11:02:34.044 MESSAGE chcanv.cpp:2312 Metrics:  m_display_size_mm: 309     g_Platform->getDisplaySize():  2560:1440
11:02:34.044 MESSAGE chcanv.cpp:2318 monitor size: 2560 1440
11:02:34.253   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Unable to connect to 192.168.1.73 on port 3000, error: OpenSSL failed - error:0A00010B:SSL routines::wrong version number
11:02:34.253   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Expecting status 101 (Switching Protocol), got 401 status connecting to ws://192.168.1.73:3000/signalk/v1/stream?subscribe=all&sendCachedValues=false&token=0, HTTP Status line: HTTP/1.1 401 Unauthorized

11:02:34.362   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Unable to connect to 192.168.1.73 on port 3000, error: OpenSSL failed - error:0A00010B:SSL routines::wrong version number
11:02:34.794   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Expecting status 101 (Switching Protocol), got 401 status connecting to ws://192.168.1.73:3000/signalk/v1/stream?subscribe=all&sendCachedValues=false&token=0, HTTP Status line: HTTP/1.1 401 Unauthorized

11:02:35.657   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Unable to connect to 192.168.1.73 on port 3000, error: OpenSSL failed - error:0A00010B:SSL routines::wrong version number
11:02:38.446   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Expecting status 101 (Switching Protocol), got 401 status connecting to ws://192.168.1.73:3000/signalk/v1/stream?subscribe=all&sendCachedValues=false&token=0, HTTP Status line: HTTP/1.1 401 Unauthorized

11:02:41.681   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Unable to connect to 192.168.1.73 on port 3000, error: OpenSSL failed - error:0A00010B:SSL routines::wrong version number
11:02:50.559   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Expecting status 101 (Switching Protocol), got 401 status connecting to ws://192.168.1.73:3000/signalk/v1/stream?subscribe=all&sendCachedValues=false&token=0, HTTP Status line: HTTP/1.1 401 Unauthorized

11:03:02.395   DEBUG comm_drv_signalk_net.cpp:183 websocket: error: Unable to connect to 192.168.1.73 on port 3000, error: OpenSSL failed - error:0A00010B:SSL routines::wrong version number
```

nohal, 2024-04-11:
Remove the "0" from the authentication token field

Hakansv, 2024-04-11:
Yes exactly! I did just found that "0" coming from ?? (My previous test runs?)
What's this token for. I may have missed some discussion?

closed 2024-04-11 by Hakansv

nohal, 2024-04-11:
It is what it says, authentication token required for connection to Signal K servers with security enabled.
The zero got there because you entered a value for the token earlier, through a bug in persistency of the connection settings causing any value in this field to change to zero.

Hakansv, 2024-04-11:
All three of my SK connections had this "0" and I'm not aware off having entered anything in any "token" control.