OpenCPN/OpenCPN #3813: Connections serial output  Apply - OK breaks the transfer

Hakansv, 2024-04-15:
**Describe the bug**
If I enable a serial output and click Apply it starts to tx. If I then click OK to close Options the output transfer stops.
If I enable and then direct click OK (without Apply) the transfer is working.

**To Reproduce**
Steps to reproduce the behavior:
1. Open NMEA Debug window and filter to e.g. COM6
2. Arrange a output serial connection to e.g. COM6
3. Click OK to close the Edit connection window.  The NMEA Debug shows the output.
4. Now disable the COM6 connection - click Apply. The output stops.
5. Enable the COM6 connection. Click Apply. The transfer starts again.
6. Click OK. The transfer stops! (Checked in both NMEA Debug window and actual traffic on COM6.)
7. Open Options->Connections. Disable the COM6 - click Apply.
8. Enable the COM6 connection. Click OK - the transfer start again.

**Expected behavior**
It should be possible to click Apply and OK after a output connection is enabled.

**Desktop (please complete the following information if applicable):**
 - OS: Checked on Windows
 - Version: Current Git master

wholybee, 2024-04-21:
I am not able to reproduce this. Testing on W11, current git master.  Using com0com to loopback com port to PuTTY running on the same machine. Both the NMEA debug window and PuTTY show the data being sent and not stopping when OK is clicked.

Hakansv, 2024-04-21:
This is a nasty one and a little special.
The bug is only valid if two connections are present to the same serial port. One disabled and one enabled.
This can happen if one use two different output filter collections and want to be able to use one or the other. Then it's of course necessary  to have control of which is enabled or not. Maybe not a common user case but still possible.
I need to edit the "reproduce-list":

Steps to reproduce the behavior:

1. Make a serial output connection to e.g. COM6 including some output filter(s)
2. Disable the COM6 connection.
3. Open NMEA Debug window and filter to e.g. COM6
4. Arrange a second output serial connection also to e.g. COM6
5. Click OK to close the Edit connection window. The NMEA Debug shows the output.
6. Now disable the COM6 connection - click Apply. The output stops.
7. Enable the COM6 connection. Click Apply. The transfer starts again.
8. Click OK. The transfer stops! (Checked in both NMEA Debug window and actual traffic on COM6.)
9. Open Options->Connections. Disable the COM6 - click Apply.
10. Enable the COM6 connection. Click OK - the transfer start again.

bdbcat, 2024-04-21:
This is a tough one.
Connections are identified generally by the port name ("COM6"), and the protocol ("NMEA0183").  Having two connections with the same port and protocol should be normally considered to be a user configuration error.
This situation is not precluded by the UI.  But the effect in real life depends upon the relative order of the two connections in the config file.  Whichever is last in the config file list will take precedence.  The display order in the UI is not considered.
Having said all that, I understand the use-case presented.  Though uncommon, it is not unreasonable for testing purposes.  Thinking....

For reference, all of this logic is contained in
`void ConnectionsDialog::UpdateDatastreams()`

bdbcat, 2024-04-21:
Patched as discussed.

Hakansv, 2024-04-21:
Works as discussed.
Many thanks.

closed 2024-04-21 by Hakansv