OpenCPN/OpenCPN #3814: Toolbar > Help  - Allow copy of the Version Number by the User

rgleason, 2024-04-16:
**Is your feature request related to a problem? Please describe.**
Make it easier for the User to Copy the OpenCPN Version Number from the Toolbar > Help.

[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

leamas, 2024-04-16:
This is actually non-trivial. The basic problem is that wxStaticText, the widget used to present text like this does not support selecting and copying text.

I have worked around this in the device dialogs like below. It works in this context where the only purpose is to display something 
which can be copied. However, it looks like crap and to use it here in the regular display would make look even worse that today due to misaligned fields, etc.

It might be possible to do something similar to the device dialogs, hiding the copyable text in the normal view but still display it when needed. I agree that being able to select and copy version and paths fill a purpose.

[image]

[image]

leamas, 2024-04-16:
While on this widget we should perhaps consider to add a "Get help" link to CF.

rgleason, 2024-04-16:
Also, perhaps easiest, only if possible.

Is there some way to click a button in the Help menu and send those files somewhere that can be accessed with the Version Number at the top?  

Just an idea.

rgleason, 2024-04-16:
Re: Get Help link to CF.    Perhaps not, as Users will just do that rather than digging a little, reading the manual and learning.  Is that perverse?

rgleason, 2024-04-16:
Also, how about a hot url link that can be copied to clipboard for   Chart path ?

rgleason, 2024-05-07:
@leamas  should this be closed?