OpenCPN/OpenCPN #3817: NMEA 0183 Filter list should be sorted alphabetically

TwoCanPlugIn, 2024-04-16:
[image]

Hakansv, 2024-04-16:
I can possible agree but this would be a future request not a bug?

TwoCanPlugIn, 2024-04-18:
I'm not sure what the objective is for the 5.10 release (I thought 5.8 was the N2K & updated comm stack), so I've taken it upon myself to identify all of the fit & finish bugs, which should never have been there in the first place. 

BTW, At the bottom of the list, there is FRPOS and CD. I assume CD is actually a talker id (Communications Device) used usually by VHF radios for DSC messages, shouldn't that be entry be DSC.

Is FRPOS some proprietary Furuno sentence ? 

And some of the filters have talker id's, most don't. In my opinion it should be sorted alphabetically and consistent (no talker id's)

Hakansv, 2024-04-19:
If you use  "Add" and enter a not valid combination, e.g. one letter, you'll see the rules and what's able to add.
FRPOS is certainty added for a reason. I don't think it's there by a change? (Some Furuno related?)
I sometimes use the filter for a talker only, like "EC", so a two letter filter is about the talker. A three letter filter is about the sentence. A five letter filter is obviously about the combination of a talker and a sentence, e.g. "ECRMC".
All described [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:connections#input_filtering):

closed 2024-05-05 by leamas

rgleason, 2024-05-05:
Hakan's link changed to  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:set_options:connections

PR #3887 (2024-05-05): connection_edit: clean up.
Some more polish to connection_edit, fix #3817.
commit 6617bbf6 (2024-05-05): Connection_edit: Sort filter sentences - #3817

This only effects presentation, the internal representation is still
unordered.

Closes: #3817