OpenCPN/OpenCPN #3821: Build of OCPN (master commit 26612dee) fails as cmake find rapidjson.h gives invalid result

jongough, 2024-04-18:
**Describe the bug**
Trying to build OCPN master using commit 26612dee fails when looking for rapidjson.h

In `(opencpn)/libs/rapidjson/CMakeLists.txt` there is a check at line 42 for rapidjson.h already being installed on the build system. This is returning an invalid result saying `rapidjson.h` was found at `/usr/include/rapidjson` however, rapidjson is not installed on the system.

**To Reproduce**
Steps to reproduce the behavior:
1. Download latest OpenCPN Master repository from github
2. Configure project
3. `rapidjson` will be found at `/usr/include`
4. No rapidjson at `/usr/include` so project will not build

**Expected behavior**
A clear and concise description of what you expected to happen.
Configuration using cmake should work on the master repository version

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: Linux mint, Kbuntu
 - Version: mint 21.3, kbunutu 23.10

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to illustrate problems, attach them if applicable.

If CMakeLists.txt line 43 changed to so that the else statement is used rather than the if statement the cmake process proceeds.

jongough, 2024-04-18:
Cleaned out all existing build information and rapidjson is now being constructed OK.

closed 2024-04-18 by jongough

leamas, 2024-04-18:
I cannot reproduce this. After removing my rapidjson package I get the following snippet when running cmake  from scratch
```
-- Using downloaded rapidjson at /home/mk/src/OpenCPN/build/_deps/rapidjson_src-src/include
-- Applying patch 0010-rapidjson-1.1.0-c++20.patch
-- Applying patch 0011-do_not_include_gtest_src_dir.patch
-- Applying patch 0012-Fix-Wclass-memaccess-warnings-errors.patch
-- Applying patch 0013-gcc7.patch

```

The general statement that cmake locates a non-existing header seems strange, that would be a really nasty cmake bug. I presume that you have tried to remove CMakeCache.txt before filing the bug (?)

leamas, 2024-04-18:
Oops,  missed a detail...