OpenCPN/OpenCPN #3829: Connections: Review typical Setup

rgleason, 2024-04-21:
Many thanks to everyone for our new user responsive Connections menu.
I need some review of this "typical setup" for inclusion in the manual.
This is not actually connected, and some experienced eyes should review as new users will be looking at it, so it should be correct.

[image]

TwoCanPlugIn, 2024-04-22:
I think there are still a few bugs and opportunity for improvement.

Unfortunate that with the new design, we have lost the inherent grouping of filters with their respective Input (Receive Data) or Output (Transmit Data) checkboxes. 

I'll give up on trying to understand why we have the option of either an inclusion or exclusion filter, nor why an empty "transmit sentences" filter, means transmit everything.

The few bugs I see at present can be illustrated in the attached screenshot.

For UDP, the outbound & inbound checkboxes are mutually exclusive. I don't know if that is the intended behaviour, but if so, they should now be radio buttons and not checkboxes. 

Duplication of Garmin Host mode option under both "General" option and on each individual connection. It is also mislabeled as "input" when displayed in an UDP Outbound connection. I'm not sure why Garmin host mode is a "general" option and when it pertains to a specific connection.  My knowledge is over 20 years out of date as I'm only familiar with the Garmin protocol on "older" Garmin devices connected via a serial/usb port. If that is still the case, then should the Garmin option only be present on TCP or UDP connections ?

Could group together in a wxStaticBox all of the autopilot related options (APB bearing precision, APB True or Magnetic bearing, APB Talker Id prefix)

[image]

leamas, 2024-05-05:
> I need some review of this "typical setup" for inclusion in the manual.

Not an issue, converting to discussion