OpenCPN/OpenCPN #3832: OpenCPNv5.9 crashing or failing to load plugin (MacOS)

antipole2, 2024-04-22:
**Describe the bug**
OCPN v5.9 refuses to update imported plugin or crashes while activating plugin.

It might only be me seeing this.  Some long while back, I imported a local development build of the JavaScript plugin that escaped detection at load time and crashed thereafter.  I think it got blacklisted in some way that has never been cleared.

Since then, I have been unable to update with a good local build.  I can import it OK but then only the Uninstall option is available - no Update etc.

If I reverted to OCPNv5.8 I could import and update the plugin and then return to v5.9 to use it OK.

However, recent builds of OCPN v5.9 (most recently 5.9.0-universal+8e73d3445) crash during the plugin loading phase of start-up, even when the JavaSCript plugin is disabled.  If I start up with all plugins disabled, it completes the start-up but in the Settings Plugins panal, ALL plugins only have the option to Uninstall.

This issue may be related to #3114 #3635

NB On Wednesday 24th April we return to Norway to start recommissioning Antipole.  I will no longer have a large screen and will mostly be relying on a roaming mobile signal, so my ability to do further investigations will be limited.

leamas, 2024-04-22:
Have you, for the sake of testing, tried to remove i manually? The paths to clear are described [here](https://opencpn-manuals.github.io/main/plugin-installer/Installation-paths.html)

leamas, 2024-04-22:
well. we have PR. If you could test this this is likely solved. But you are the only one which can reproduce the problem so far, I have not seen any similar reports. 

If I understand the crash dump correctly you should have messages in the logfile about which plugin it tries to load when crashing. This requires that you start opencpn using `--loglevel debug`

bdbcat, 2024-04-22:
So, lets go carefully here.  If we scare the bug away, we may not see it again until it comes out of hiding.
The thing to do is manually remove all plugins from the library location, one-by-one, testing at each stage.

antipole2, 2024-04-22:
I removed the plugins manually one at a time - both the dylib and the relevant shared support folder.
Once I had removed VDR, OCPN started OK.
I reinstalled from the catalogue and it is working.

My original problem following that rogue build is supposed to have been fixed.  I was still seeing a problem and thought the fix was not working for me.  It now looks like it was a bad copy/version of VDR.  This now seems OK.

Apologies for taking time on this.

But I note:  if I crash OCPN and then start with "Yes, disable plugins", the only option is to uninstall them.  I see no option to enable them.

bdbcat, 2024-04-22:
Tony...
Good.
As designed.  For users, the only reasonable option is to uninstall all plugins.  Then restart and add one-by-one until failure.
And really, our goal is to never publish a plugin that crashes on startup.  This may mean more effort for devs in rapid dev cycle we use.  But that is our job.

bdbcat, 2024-04-22:
Please close if satisfied.

antipole2, 2024-04-22:
Hereforto, if I had established plugins installed and my own under-development-import crashed, I could start without plugins enabled, import a fixed version of my plugin and then re-enable the others.

So now I need to manually remove my rogue plugin, restart and import a fixed version.
Else I have to remove all plugins and start again.

Can live with that if that is how it is.
Tony

bdbcat, 2024-04-22:
That is the process that I use.  Except I do not bother with Import.  During normal debug, I don't really care about the integrity of the catalog.  I'm working on the plugin itself.
I just copy the .so/.dll/.dylib file to the proper location manually, and start OCPN.

antipole2, 2024-05-26:
Reopening to report a discovery.  will close again in ten days.

A more recent OCPN build gave import error message when I tried to import a local build
“Error extracting metadata for tarball”

I have been aware that if I import under 5.8 successfully, the version number appears as the old one until I do a remote build, after which it is correct.

So… I deleted the CMakeCache.txt file in build-osx and now it imports OK under 5.9.

Lesson: when editing Plugin.cmake to, e.g. bump version number, **remove the CMakeCache.txt file**.

antipole2, 2024-06-08:
Closing as no comments

closed 2024-06-08 by antipole2