OpenCPN/OpenCPN #3835: ShapeBaseChart::DoDrawPolygonFilledGL performance

nohal, 2024-04-22:
For high resolution basemaps we can't tesselate the whole polygons as the impact  on performance is way too high. The points outside the viewport have to be filtered out.

When returning over a different edge of the bounding box, some artificial point(s) have to be added to ensure correct coverage.

nohal, 2024-04-22:
Something like this:

```diff
diff --git a/gui/src/shapefile_basemap.cpp b/gui/src/shapefile_basemap.cpp
index 2301f107b..70d01731a 100644
--- a/gui/src/shapefile_basemap.cpp
+++ b/gui/src/shapefile_basemap.cpp
@@ -342,7 +342,6 @@ void ShapeBaseChart::DoDrawPolygonFilledGL(ocpnDC &pnt, ViewPort &vp,
       vp.GetBBox().GetMinLon() <= -180 || vp.GetBBox().GetMaxLon() >= 180;
   auto polygon = static_cast<shp::Polygon *>(feature.getGeometry());
   for (auto &ring : polygon->getRings()) {
-    size_t cnt{0};
     GLUtesselator *tobj = gluNewTess();
 
     gluTessCallback(tobj, GLU_TESS_VERTEX, (_GLUfuncptr)&shpsvertexCallback);
@@ -356,9 +355,27 @@ void ShapeBaseChart::DoDrawPolygonFilledGL(ocpnDC &pnt, ViewPort &vp,
 
     gluTessBeginPolygon(tobj, NULL);
     gluTessBeginContour(tobj);
+    bool lastin = false;
+    shp::Point lastpoint = ring.getPoints().front();
+    if (vp.GetBBox().Contains(lastpoint.getY(), lastpoint.getX())) {
+      lastin = true;
+    }
     for (auto &point : ring.getPoints()) {
-      AddPointToTessList(point, vp, tobj, idl);
-      cnt++;
+      if (vp.GetBBox().Contains(point.getY(), point.getX())) {
+        if (!lastin) { //Last point outside of the viewport before getting back to visible part must be included
+          AddPointToTessList(lastpoint, vp, tobj, idl);
+        }
+        AddPointToTessList(point, vp, tobj, idl);
+        lastin = true;
+      } else {
+        if (lastin) { // First point outside of the viewport must be included
+          AddPointToTessList(point, vp, tobj, idl);
+        } else {
+          // TODO: If we got to a different edge of the bounding box, we need to add an artificial point in the corner
+        }
+        lastin = false;
+      }
+      lastpoint = point;
     }
     gluTessEndContour(tobj);
     gluTessEndPolygon(tobj);
```

bdbcat, 2024-04-22:
Tested?

nohal, 2024-04-22:
Yes, but it is missing the TODO part of course, so not ready to be merged.

bdbcat, 2025-04-26:
OBE, O5.12 significantly reworked shapefile rendering.
Close?