OpenCPN/OpenCPN #3836: First start Wizard. Previous AIS defaults and plugins are missing.

Hakansv, 2024-04-22:
First time test on clean Win PC. (No test fake)
Code tested is today's Git master.

Great:
Units are set as defined in the Wiz. (I miss wind speed. PR to come.)
All four serial USB was found with correct baud rates.
Charts was installed as listed in the Wiz.
O started. Charts installed and ship position was directly found. 

Issues:
1. No plugins are enabled. I'd expected at least WMM and maybe Dashboard.
2. Nearly no Ships AIS target preferences are set, See pict.. We had some defaults before. They where good for a start but can be discussed.
3. I suggest we set "Display Category" to "All" as default? Most devices would manage that today. New users often miss that.

[image]

TwoCanPlugIn, 2024-04-22:
One observation that needs confirmation.

If you select the chart folder(s) during the installation of OpenCPN, they don't seem to appear in the First Start Wizard ?

Hakansv, 2024-04-22:
Appear?
You have to click the "Open" or what it says to open file explorer to point a chart catalog.

TwoCanPlugIn, 2024-04-22:
I think I haven't explained myself well.

See the following attachments. If you add a chart folder during **install**, they don't seem to appear in the first start wizard.
[image]
[image]
[image]

Hakansv, 2024-04-22:
Hmm. That "Add chart directory" in the "Ready to install" window I have not seen. Maybe missed but not sure. I'll check tomorrow.

bdbcat, 2024-04-23:
I think I agree.
@TwoCanPlugin     What do you think?

TwoCanPlugIn, 2024-04-23:
Yes, that makes sense, 

If it is only a Window feature in the NSIS installation, then do away with it.

All users will then have a consistent way to add charts in from the First Start wizard.

Hakansv, 2024-04-23:
Very good!
@bdbcat  I've tried to look into NSIS.template but to do this tweak is over my head. So please you or anyone else...
But while there I saw we need to exclude XP since long. This was not complicated so before editing the template maybe merge this PR #3839 if OK. 

Now when this little digression to the main issue is finalized the AIS default is next on table?

rgleason, 2024-04-23:
Hakan wrote: "No plugins are enabled. I'd expected at least WMM and maybe Dashboard."
We want New Users to get Charts easily (at least some sample charts)
Why not "Enable" Chartdownloader_pi" during initial configuration?
It would also be helpful to Install and Enable O-charts_pi for purchase of charts, but that is more of a problem.

bdbcat, 2024-04-24:
Negative on o-charts as default load.
Users should master at least some OCPN basics before loading o-charts.  The issues of licensing, dongle (or not), system names, etc. are all very distracting, and beyond  a complete novice grade.
Never mind that a relatively small percentage of users even need o-charts.
Less is more for novices.

rgleason, 2024-04-24:
What about enabling Chartdownloader_pi?

Hakansv, 2024-04-25:
fixed by 99b528f98

closed 2024-04-25 by Hakansv