OpenCPN/OpenCPN #3838: Lighthouse sector artifacts a

leamas, 2024-04-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

bdbcat, 2024-04-23:
??

leamas, 2024-04-23:
hm... after restarting OpenCPN I cannot reproduce this. The screenshot is evidence that it could happen, but I seem to have missed something in the "To Reproduce". 

Leaving open for now, but unless it happens again or someone else chimes in I will close in a few days.

leamas, 2024-04-26:
This has not happened again. Closing

closed 2024-04-26 by leamas