OpenCPN/OpenCPN #3840: S-57 objects disappear when panning with the mouse (addition to #3350)

bc38, 2024-04-24:
[Video](https://www.dropbox.com/scl/fi/78sxokctjr9kr1h5xr1cu/2024-04-24-130312.mp4?rlkey=cxawiro8kqiznxpfvf9plbj6s&st=p3zlajf0&dl=0)
There is no problem if you move the map using the arrow keys on the keyboard.
There is a problem if you move the map up with the mouse; when moving left, right, or down, the problem does not reproduce.

Hakansv, 2024-04-24:
I can occasionally see something similar.
That's when two or more chart files are present for the location. Then when moving the chart with the mouse the quilt logic sometimes show a chart file with smaller scale without soundings. If I change zoom level or test to increase the "Vectorchart zoom/scale slider" one step for the same zoom level the effect is not there.
(You know the "Vectorchart zoom/scale slider" can be easily adjusted if you key "D")

bc38, 2024-04-24:
> I can occasionally see something similar. That's when two or more chart files are present for the location. Then when moving the chart with the mouse the quilt logic sometimes show a chart file with [...]

No, it's definitely not about quilting. In the last example there is one map cell. When scaling, everything is displayed correctly. The problem only occurs when the mouse pans the map to the top of the screen

bdbcat, 2024-04-26:
Any change after this latest commit?

bc38, 2024-04-26:
> Any change after this latest commit?

Without changes!
Let me clarify that in addition to depth marks, isobath polylines, coastlines, and relief elevation lines are also used.

bc38, 2024-05-27:
Dave, do you have any ideas on how to fix this?

bdbcat, 2024-05-27:
Not at the moment.  I have had some trouble to duplicate the problem.
Anyone else experiencing this effect?

bc38, 2024-05-27:
I need to make an announcement on the forum

bdbcat, 2025-04-26:
Closing, please reopen if needed for O5.12

closed 2025-04-26 by bdbcat