OpenCPN/OpenCPN #3843: Bundled plugins manuals: strange localtion

leamas, 2024-04-25:
**Is your feature request related to a problem?**
If/when trying to maintain the bundled plugin manuals, the sources are in obscure locations in the opencpn plugins manual project at https://github.com/opencpn-manuals/plugins.  

**Describe the solution you'd like**
Store the bundled plugins manual together with the plugin, in the same way as almost all external plugins.

**Describe alternatives you've considered**
We could leave in in place, perhaps just dropping a README pointing to the current location

**Additional context**
Basically a no-brainer besides that this will add some weight to the repository -- the manual is both text and images.

rgleason, 2024-04-25:
Who wrote this?  Not me.
I just think we should have a download (pdf?)  from the User Manual, of the Plugin User Manual for offline use.
Is this possible?

leamas, 2024-04-25:
Could you please try to stay on subject i. e., that the sources for the bundled plugins are in a strange location.

Just once?

Please?

leamas, 2024-04-26:
No, this sheer size should not be a problem. The PR adds about 3MB. Given that the size of the repo is measured in GB the 3 MB added here should not be a problem

PR #3844 (2024-04-26): Add plugin manuals to main tree
As heading says: move the bundled plugin manuals to the bundled plugins, storing the manual in the same way as for other plugins.

Closes: #3843
closed 2024-04-26 by leamas