OpenCPN/OpenCPN #3847: Dashboard icon does not work when the tactics dashboard plugin is active

bc38, 2024-04-26:
[image]

When you disable the plugin, the tactics dashboard and reboot the CPN icon is restored

rgleason, 2024-04-26:
No. Tactics uses older Dashboard code, although Dashboard has been enhanced significantly. 
Therefore you cannot run both Dashboard and Tactics at the same time!
Not a good idea, and don't expect it to change.

WONTFIX  not a bug
Will the original poster please close this.

leamas, 2024-04-28:
Well... saying the this is "not a bug" and WONTFIX is an obvious contradiction. Either it isn't a bug, or it's a bug that will not be fixed. It cannot be both.

That said, **if** the root cause is that Tactics uses the same section as Dashboard in the config file (which, BTW,  is called opencpn.ini only on Windows) this is most likely a bug in the Tactics plugin and should be filed against it. 

The Tactics repo is at  https://github.com/rgleason/dashboard_tactics_pi.

I cannot see that this report is about core opencpn in any way, though.

rgleason, 2024-04-28:
WONTBEFIXED is the answer.  User can learn not to run both.

leamas, 2024-04-29:
OK, this is a Tactics bug which will not be fixed. 

I will file a new bug against Tactics and close this one, effectively transferring the bug to tactics.  The tactics bug could be closed be you in your capacity as maintainer for this plugin.

closed 2024-04-29 by rgleason