OpenCPN/OpenCPN #3848: API 1.17 Creating a route does not allow setting speed or route colour

Rasbats, 2024-04-26:
**Is your feature request related to a problem? Please describe.**

Using API 1.17/PlugIn_Waypoint_Ex/PlugIn_Route_Ex it is not possible to define the speed for the route or the colour of the route.

**Describe the solution you'd like**
When the next API is added make these features possible.

**Additional context**
Creating a route using API 1.17 code:

[image]

Importing a route using the Route Manager:

[image]

Rasbats, 2024-04-26:
[https://github.com/Rasbats/sar_pi/issues/9](https://github.com/Rasbats/sar_pi/issues/9)

Rasbats, 2024-10-28:
My mistake. Needs some more coding in the plugin.

Rasbats, 2024-10-28:
The Plugin API is the problem.  Still needs a fix.

rgleason, 2025-06-08:
@Rasbats Mike, does API 1.18 solve the problem?  Can this be closed now?

Rasbats, 2025-06-08:
imo API 1.18 will not solve the problem. No property for the colour of a route when coding a route.

rgleason, 2025-06-08:
Well, as a practical matter, I kind of doubt any API change will be made to 1.17 at this point.  Perhaps you want to re-title this?