OpenCPN/OpenCPN #3852: AIS target list. Targets are not deleted from the list when removed from screen.

Hakansv, 2024-04-27:
I'm not sure if this is intentional or a bug. Thus not classified a bug.
This is also a test scenario to emphasize but the main q is there.
The AIS target list have in my example received not so few targets from one source, pict 1.  The AIS target deletion time out is set to 8 minutes and after 15 minutes all targets from that source are also deleted. But all are still listed in the AIS target list. Pict 2.

I would have expected the target list to be updated not only when new targets are received but also when targets are timed out from screen?
Or are there some thoughts about keeping them on the list "forever"?

Pict 1
[image]

Pict 2
[image]

bdbcat, 2024-04-27:
I would call this a bug.

bdbcat, 2024-05-05:
Not really a bug, but unexpected behavior.
1.  The AIS target list is updated asynchronously every 5 seconds from main OCPN timer loop.
2. For some reason, I cannot remember why, we do not delete "lost" or "display-removed" targets from the list until 3X the specified "display-removed" interval selected.  This would be 45 minutes in your config.

```
        //      If we have not seen a static report in 3 times the removal spec,
        //      then remove the target from all lists
        //      or a lost ARPA target.
        if (target_static_age > removelost_Mins * 60 * 3 || b_arpalost) {
          xtd->b_removed = true;
          plugin_msg.Notify(xtd, "");
          remove_array.push_back(xtd->MMSI);  // Add this target to removal list
        }
 
```

Probably we are trying to retain the target static data in the list, so that returning targets get full info available sooner.
But I agree, with the default timeout values, 45 minutes is just too long.

Thoughts on all this?

Hakansv, 2024-05-06:
> returning targets get full info available sooner

What would that be?  Old position, SOG, range for example are of no use. Ship name is already available from the name hash. The procedure to add a target in the list again would not be a performance issue using today's computers.
There could be a nice to have history to see lost (buddy?) targets on the list but since it's long time lost the list item is not really usable. 
To delete a lost targets from the target list after not more than two times "g_RemoveLost_Mins", normally 16 to 20 minutes may be sufficient? Personally I see no use of old data and it could be deleted also from target list when it's lost from screen.

bdbcat, 2024-05-07:
All valid points.  This may date way back to before the AIS name cache was implemented.
Anyway, implemented as described:  "deleted also from target list when it's lost from screen."
Ready for close.

closed 2024-05-07 by Hakansv

aviatorhh, 2024-05-07:
I know it is closed but my 5 cents from a safety perspective is that a target may be in distress and therefore you might have its last known position in that list. It is a rare case but something to think of to make the timeout configurable maybe.