OpenCPN/OpenCPN #3857: Compile guide on windows is outdated

Kitiara, 2024-04-28:
Hello,

The compile guide [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows) is outdated due to wxWidgets.

- The `cd build\msw` part should be `cd build/msw` and also the nmake parts are not working due to `stdio.h` not found.
- The alternative way is not working either, due to missing binaries.
- I have tried `buildwin/win_deps.bat` and that too doesn't seem to be working because cmake fails at the end.

OS: Windows 11

nohal, 2024-04-28:
The current developer manual is here: https://opencpn-manuals.github.io/main/opencpn-dev/windows.html

closed 2024-04-28 by nohal

Kitiara, 2024-04-28:
FYI once the wget and 7z is installed by buildwin\win_deps.bat, Visual Studio 2022 Tools Command Prompt didn't get the environmental variables refreshed. So i had to retry with a second command prompt for a second time.

rgleason, 2024-04-28:
As it says,  **OLD 2021 -User Manual**