OpenCPN/OpenCPN #3863: UDP in/out connection bug

Corsair-63, 2024-04-28:
Raspberry Pi 5 - 8GB
Enclosure Neo with NVME SSD connected to PCI connector.
Windows 11 ARM64 up and running ver 22631.3527
O Version 5.9.0-0+10912b6 Build 2024-04-26

no problem with LAN (via USB adaptor) and another devices.

setup the data connection via UDP with the multiplexer, received all data.

also tried to make in/out the connection as my other computer but not able to set both at the same time.
when I click on input the output comes off, viceversa when I click on output the input comes off, could not have both at the same time, and immediately the IP changes to 0.0.0.0
[image]

[image]

Why the "N" before the 0183?
[image]

Also tried in same O version but windows 11 x64, same behaviour.

but in another computer that was set previously of this O version the up/down is available, same os and OCPN (same version and architecture ARM64) with the computer on board Apcsilmic (Qualcomm PC for Windows,  8 GB RAM too).
Up/Down enable without any problem.
[image]

Hakansv, 2024-04-28:
> when I click on input the output comes off, viceversa 

"It's not a bug it's a feature" The radio button was implemented to not allow the risk for a endless feedback. 

N0183 is a obviously a abbreviation for NMEA 0183. Like N2000 is for NMEA 2000 or N2k. Lack of space and meaning the may be clear in this context.

leamas, 2024-04-28:
No bug to be found here, but perhaps some misunderstandings. Converting to discussion