OpenCPN/OpenCPN #3865: The Transmit Filter Selection List has all _selected_ standard sentences _appended_ to the list.

torrmundi, 2024-04-28:
**Describe the bug**
The Transmit Filter Selection List has all _selected_ standard sentences _appended_ to the list.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options>Connect>Data Connections>Edit Connection
2. Check "Output on this Port and in Output Filtering section, click "..."
3. Click Select All, then OK. All the selected sentences are added to the Filter line.
4. Again, click on "..." and scroll down the list of sentences.
5. You'll find all the sentences on the Filter line have been appended to the list of sentences.

**Expected behavior**
Only the non-standard sentences that are *Added* to the list should be appended to the list of sentences. Standard sentences that are simply enabled should not be appended.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Win11 Home
 - OpenCPN Version 5.9.0

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

rgleason, 2024-04-29:
John, I assume you are using the most recent version?  Just checking...

torrmundi, 2024-04-29:
Uh, no I was using opencpn_5.9.0-0+1336.6fe2666_setup.exe.
I've just installed the latest I found on Cloudsmith, opencpn_5.9.0-0+1502.e0d43cb_setup.exe.
It seems to fix the bug. I'll update Github.
I also have to make new screenshots of the Add / Edit Connections screen, which has changed.
Thanks,
John

rgleason, 2024-04-29:
John if this is Issue is resolved, it is now the responsibility of the OP (original poster) to CLOSE it. Thank you for posting..

torrmundi, 2024-04-29:
Fixed in opencpn_5.9.0-0+1502.e0d43cb.

closed 2024-04-29 by torrmundi