OpenCPN/OpenCPN #3866: Long exit delay causes "Do you want to terminate appliation" dialog

leamas, 2024-04-29:
**Describe the bug**
I have since long observed a rather log delay from closing opencpn until it actually exits. Now, on a RPi clone running XFCE I get the dialog below which makes me file this.

**To Reproduce**
Steps to reproduce the behavior:
1. Start opencpn
2. Configure two serial connections, one input, one output (not clear to me if load affects this)
3. Click X in top-right window corner
4. Wait around a second
5. See dialog below

**Expected behavior**
There should be no delay

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
```
$ uname -a
Linux homsan 6.1.43-rockchip-rk3588 #1.1.8 SMP Fri Feb  2 21:16:10 CST 2024 aarch64 GNU/Linux
```
That is, Debian 12/Bookworm updated as of "recently" + Rockchip's kernel fork 

**Smartphone/Tablet (please complete the following information if applicable):**

Orange PI 5 (similar to RPi 5, but  based on RK3588S -> faster), NVME disk, 8Gb
 
**Additional context**

Disabling the serial drivers makes the problem go away.

leamas, 2024-04-29:
Too late for 5.10? Might have to wait until we something about  #3116.

leamas, 2024-04-29:
> .Unless the fix is super simple I think it can wait.

Agreed

nohal, 2024-04-29:
Maybe mention `gsettings set org.gnome.mutter check-alive-timeout 60000` or even `gsettings set org.gnome.mutter check-alive-timeout 0` somewhere. The default 5 seconds must be a pain with pretty much anything on the ARM SBCs...

leamas, 2024-04-29:
hm... This is on X11, no  Wayland or mutter. As a quite large part of the Arm users, I guess.

leamas, 2026-09-08:
On current master 3784ed1c7 I cannot reproduce this any more. It used to be about serial load, the connections below includes a 0183/9600 bps  connection so quite substantial. Screenshot from my Orange pi 5, roughly similar to RPi 5.

One possible difference is that my OS is Armbian, basically vanilla Debian 13 with a patched kernel. As such, it uses wayland instead of X11. 

Situation basically the same when building 5.14.0. Since I cannot reproduce any more marking as wontfix. Unless there is more input I will eventually close this bug.

[image]

leamas, 2026-09-08:
Adding info-needed: Willl close unless there is more input.

Hakansv, 2026-09-08:
Never seen on my RPis. Although on all Debian 12 and 13 I use Wayland + labwc

closed 2026-09-13 by leamas