OpenCPN/OpenCPN #3870: User Manual - Questions - Quck Start

rgleason, 2024-04-30:
We are trying to get the Quick Start section finished.

[Serial + Nmea 0183 ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183)  Only has information about connecting Flatpak. Is it correct?  Is it located in the right page?   Can it be deleted?

There are also two sub-pages 
[Linux](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183:linux) Which is about Linux device access, fixed device names and testing.  Can it be deleted?

[USB and Serial ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183:serial_usb) Which also about Linux, Flatpak and udev. Can it be deleted?

I would like to get these completed or deleted or to join all these pages into the main page, or delete the sub-pages.

What other information is needed for 
[Serial + Nmea 0183 ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183)  as it only seems to reference Linux and Flatpak.

leamas, 2024-04-30:
> [Serial + Nmea 0183 ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183)

Delete it

> There are also two sub-pages
[Linux](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183:linux) Which is about Linux device access,

Delete it

> [USB and Serial ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183:serial_usb)Which also about Linux, Flatpak and udev.

Delete it

>  [Serial + Nmea 0183 ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183)as it only seems to reference Linux and Flatpak.

Delete it

Of course, deleteing these "manual" instructions cannot be done until the work I submitted in the issue yesterday is merged first.

closed 2024-05-02 by rgleason

leamas, 2024-05-02:
#3868 should not be considered here, it's just some polish, there is no need to block the manual on it.

rgleason, 2024-05-02:
OK, but are these pages correct now?  Quite frankly, I've forgotten if this has been reviewed.

We are trying to get the Quick Start section finished.

[Serial + Nmea 0183 ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183)Only has information about connecting Flatpak. Is it correct? Is it located in the right page? Can it be deleted?

There are also two sub-pages
[Linux](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183:linux) Which is about Linux device access, fixed device names and testing. Can it be deleted?

[USB and Serial ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183:serial_usb)Which also about Linux, Flatpak and udev. Can it be deleted?

I would like to get these completed or deleted or to join all these pages into the main page, or delete the sub-pages.

What other information is needed for
[Serial + Nmea 0183 ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183)as it only seems to reference Linux and Flatpak.

leamas, 2024-05-02:
> Serial + Nmea 0183  

Delete it.

> Linux device acess...

Delete it

> USB and Serial 

Delete it 

> Serial + Nmea 0183 

Duplicate

leamas, 2024-05-02:
The only relevant info is what I submitted somewhere

leamas, 2024-05-02:
... i. e. what I wrote [here](https://github.com/OpenCPN/OpenCPN/discussions/3776#discussioncomment-9262067)

rgleason, 2024-05-02:
Alec, ok I've deleted those pages, and repurposed  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:serial_n0183
for your 2 screenshots and text.

Where exactly should this information be located in your opinion?  If this page an ok location, is the title OK?
Thanks.  Sorry about all the questions.