OpenCPN/OpenCPN #3871: User Manual - Quick Start - SignalK page

rgleason, 2024-04-30:
Is this page finished?

[SignalK](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:signalk)

See the notes at the top please. 
Provide direction please.

TwoCanPlugIn, 2024-05-01:
In terms of a very brief guide, it's fine.

The only suggestions I have is to be more explicit for users running SignalK & OpenCPN on the same PC and users running on separate machines.

Eg. If you are running SignalK on a separate PC, you must enter its address in the SignalK Connection settings.

If you are running SignalK Server and OpenCPN on the same machine (for example if using OpenPlotter), it is easier to use "localhost"

I personally would prefer them to use "localhost" rather than the IP address 0.0.0.0. They may find it confusing reading other documentation or if configuring an inbound UDP connection using the address 0.0.0.0. 

In that context it means INADDR_ANY. (ie. listen on all networks) whereas in this context, connecting to 0.0.0.0 means connect to 'this host' on 'this network' and generally it is equivalent to "localhost"

rgleason, 2024-05-01:
Thanks Alec. I've added your suggestions as a FIXME at the top.

leamas, 2024-05-01:
>In that context it means INADDR_ANY. (ie. listen on all networks) whereas in this context, connecting to 0.0.0.0 means connect to 'this host' on 'this network' and generally it is equivalent to "localhost"

The argument to use "localhost"  is actually stronger than that. The IP address of "this host" a k a "localhost" is 127.0.0.1. If OpenCPN actually uses "0.0.0.0" for "this host" we should avoid  using that, it's  confusing for anyone with basic ipv4 knowledge. Using "localhost" is the simple fix.

Hakansv, 2024-05-01:
The 0.0.0.0 is now used for any "localhost" default in Git master and works for all system I've tested. 
The manual note reflects master code. Until changed the manual would reflect that I suppose?

For UDP output we can also use e.g. 192.168.x.x also quit strange for some?

rgleason, 2024-05-01:
"For UDP output we can also use e.g. 192.168.x.x   also quite strange for some?"

So, for UDP output using IP Address: 0,0.0.0  broadcasts to all devices on that network and so does 192.168.x.x?

Hakansv, 2024-05-01:
For UDP **output** you can't use localhost or 0.0.0.0.

rgleason, 2024-05-02:
Should I close this now?

leamas, 2024-05-02:
Do  you open issues which you  cannot understand if they are resolved or not?

If this indeed is the fact I suggest that you close this...

rgleason, 2024-05-02:
I don't know signalK,

closed 2024-05-02 by rgleason