OpenCPN/OpenCPN #3872: Filenames and German Umlaut

jongough, 2024-04-30:
Describe the bug
.gpx files containing german umlaut in filename / title cannot be saved or oadad under android.
Save routes with umlaut like ä, ö or ü in filename are empty
Files saved under windows containing umlaut cannot be loaded for example from draw plugin.

To Reproduce
Steps to reproduce the behavior:

    create route
    give name containing ä, ö or ü
    export route with a filename containing ä, ö or ü
    file is empty

or with draw plugin

    create area on windows
    give title containing ä, ö or ü
    export are with od plugin
    import are on android with od plugin

Expected behavior
Should work with national. characters, if not a warning dialog appears, that filename does not work.

Smartphone/Tablet (please complete the following information if applicable):

    Device: Galaxy Tab A 2019

Additional context
[Test German Umlaut ä.zip](https://github.com/OpenCPN/OpenCPN/files/12138384/Test.German.Umlaut.a.zip)
[Draw Example.zip](https://github.com/OpenCPN/OpenCPN/files/12138385/Draw.Example.zip)

Apeared first with od plugin, so reported https://github.com/jongough/ocpn_draw_pi/issues/530

See: [https://github.com/OpenCPN/OpenCPN/issues/3363#issue-1817057091](https://github.com/OpenCPN/OpenCPN/issues/3363#issue-1817057091)

The issue was closed without explanation. The problem still exists in OCPN when run on Android. See [https://github.com/OpenCPN/OpenCPN/issues/3363#issuecomment-2081667523](https://github.com/OpenCPN/OpenCPN/issues/3363#issuecomment-2081667523) where it is described in which part of OCPN the issue exists. 

This needs fixing not closing without explanations

bdbcat, 2024-05-01:
Two considerations:
1.  Umlaut in filename:  This is a system issue, probably related to wxWidgets file selector dialog.
2.  Umlaut in GPX file contents, e.g. "Title, Name":  This relates to XML parser library used by OCPN.

May I understand that this entire issue (both  parts) relates to Android only?

jongough, 2024-05-01:
I have tried the ODraw gpx file both with and without the umlaut in the filename. I did not change the content at all. With the umlaut the file does not load, without the umlaut (just renamed the file to remove it) it did load and displayed correctly.

I have tried importing the gpx files supplied above on a ubuntu system and it works fine, but on an Android phone it does not complain but nothing happens to the contents. I tried debugging it on the ubuntu system and could see where it went to do the file access and that is where I saw the android only bits of code that I think are not working. I don't have an android system I can run debugging on to see which bit actually fails, so cannot narrow it down any further.

So it would appear to just be the filename processing.

bdbcat, 2024-05-01:
OK, thanks.
Will debug on Android, as time allows.