OpenCPN/OpenCPN #3873: NMEA 2000 on Serial?

rgleason, 2024-05-01:
**Describe the bug**
Serial ports (RS232/RS422) are not part of the NMEA 2000 standard.
NMEA 2000 devices do not directly connect to serial ports; they use the NMEA 2000 backbone.

[image]

BTW,  I think allowing both Receive Input and Output on one Connection is complicating setup when two explicit connections can be made, which will show more clearly in the menu.

bdbcat, 2024-05-01:
N2K serial connections are used to connect to various N2K<->Serial gateways, such as Actisense NGT-1, and some Yacht Devices gateways.
These gateways are complimentary to various N2K<->Ethernet gateways,supported in OCPN by an N2K network connection.

NOTABUG

rgleason, 2024-05-01:
These Devices use this configuration?

- Actisense NGT-1 NMEA 2000 to USB Interface:
- The Actisense NGT-1 is a versatile USB interface that allows bidirectional communication between your PC and the NMEA 2000 network.

- Yacht Devices NMEA 2000 USB Gateway:
- The Yacht Devices NMEA 2000 USB Gateway connects to your PC via a USB cable.

- TwoCan Plugin and various Canbus?

closed 2024-05-01 by rgleason