OpenCPN/OpenCPN #3874: SignalK  changed to use Localhost rather than 0.0.0.0

rgleason, 2024-05-01:
**Is your feature request related to a problem? Please describe.**

Please see this note written by Alec   https://github.com/OpenCPN/OpenCPN/issues/3871#issuecomment-2087759106 suggesting the use of Localhost instead of 0.0.0.0

[image]

TwoCanPlugIn, 2024-05-01:
Rickt,

This is not a bug or an enhancement, merely a suggestion for the manual.

However that being said, what would be useful would be to provide a hint in the text control using the control's SetHint function.

There may be sufficient room in the control for something like "If SignalK is on same PC use localhost, else enter IP address"

rgleason, 2024-05-01:
See what Alec says here  https://github.com/OpenCPN/OpenCPN/issues/3871#issuecomment-2088173899
He says argument for using Localhost is stronger.  I really don't know.  Just to understand it better:

Localhost IP Address -  A special hostname that allows a computer to connect to and communicate with itself. The corresponding IP address for localhost is 127.0.0.1.   "Host typically refers to a server on the itnerent.  Localhost represents a computer that is making a request to itself, in this case acting as a virtual loopback server.

127.0.0.1  IP Address - A reserved address for local servers on computers.  Localhost and 127.0.0.1 are interchangeable.

0.0.0.0  IP Address  - A non-routable placeholder address, a catch-all and a default route address depending on context. In some cases it means accept all IP addresses (incoming) in other cases it may mean block all IP addresses and it can represent the "default route" when configuring network interfaces.

rgleason, 2024-05-01:
@TwoCanPlugIn 
Steve, are you referring to the "SignalK Server"
There may be sufficient room in the control for something like "If SignalK is on same PC use localhost, else enter IP address"

TwoCanPlugIn, 2024-05-01:
Rick,

Yes, this is purely for a user configuring a connection to a SignalK Server.

The suggested hint "If SignalK is on same PC use localhost, else enter IP address" is intended for the IP address field for a SignalK connection.

SignalK connections using Web Sockets, (which is what OpenCPN uses) are bi-directional, meaning that the UI does not have a separate receive & transmit option which is quite different to the other OpenCPN transport protocols. 

However most users just want to receive data from SignalK so no need to mention transmitting to SignalK in the manual.

The simplest and most unambiguous term to use would be "To configure a connection to a SignalK Server do this..." 

Do not confuse the issue by saying "to configure a receiving input connection"

leamas, 2024-05-02:
If something like this needs a lot of explanations, it's sign that it might be better handled in code.

The "signalK websocket server on localhost" use case should be easy to autodetect and just have "localhost" as a pre-defined value. Simple  enough to land before 5.10.

For other use cases auto-detecting is possible in several use cases covering a large share of users. However, it is as I understand it not really possible in the general case. Perhaps it would be easier to just add "Check" button which verifies that the  value entered by user indeed is a signalK server.

Hakansv, 2024-05-02:
> "Check" button which verifies that the value entered by user indeed is a signalK server.

That function are more or less there and in the proposed manual text: Click "Show NMEA Debug window" (That would be changed to "Show Data Debug Window" or similar)

leamas, 2024-05-02:
I mean something simpler which just returns a popup saying things like "192.168.12.24" is a valid signalK server" or "Cannot find a signalK server at 192.168.12.24".

The NMEA Debug window should IMHO not be an integrated part of a basic setup procedure, new users needs time to accommodate to  things like NMEA message types, etc.

Hakansv, 2024-05-07:
This would have been solved by a9b69e0e5d ?

closed 2024-05-07 by rgleason