OpenCPN/OpenCPN #3875: User Manual - GNSS Install - Are Flatpak and RPI the same a Linux?

rgleason, 2024-05-01:
Do Flatpak and RPI need unique GNSS Install descriptions?

Corsair-63, 2024-05-01:
I have a Raspi4 with Raspbian and from time to time I test the connection with my Android 14 pixel 8 Pro to have a GNSS backup (Bluetooth), while I have not any problem with windows devices because immediately, I switch on the app "Bleutooth GPS output" the computers start receiving the signal.
but for Raspbian on every time that connects to the RPi4 it exchanges the channel and must do the full process of connection again and 40% of the times not able to connect.
for sure that USB GNSS would be much easier, but BT is a 'pain in the ....'

in terminal, most of the times as 'su':
**$ su
Contraseña:** 

**rfkill list**
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

**hcitool scan**
Scanning ...
	XX:XX:XX:XX:XX:XX	Pixel 8 Pro (copy the MAC of your GNSS BT device)

**sdptool browse XX:XX:XX:XX:XX:XX**   <---- with the pasted MAC above
Browsing XX:XX:XX:XX:XX:XX ...
....
Service Name: SPP slave
Service RecHandle: 0x10013
Service Class ID List:
  "Serial Port" (0x1101)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: nn    <--- This channel
Profile Descriptor List:
  "Serial Port" (0x1101)
    Version: 0x0102

**sudo rfcomm bind /dev/rfcommY XX:XX:XX:XX:XX:XX nn**  (the Mac pasted above and the SPP slave channel number in last command)

**rfcomm**
rfcommY: XX:XX:XX:XX:XX:XX channel nn connected [tty-attached] 

(this will give you the device to connect in OCPN rfcomm0, 1, ...)
(Y is the number of rfcomm you want to setup 0, 1, 2 ...)
and setup in OCPN, com port as 4800:
[image]

leamas, 2024-05-02:
> Do Flatpak and RPI need unique GNSS Install descriptions?

Flatpak: yes. 

RPi: No, it's either Debian or Flatpak

leamas, 2024-05-02:
... and please stop abusing the issues mechanism for these kind of questions.  If you cannot answer the basic questions for an issue (describe the problem, steps to reproduce, etc.) it's not an issue but a discussion.

rgleason, 2024-05-02:
@leamas  Alec can you please convert to a Discussion? Thanks.  I've added a note about Flatpak in connect GNSS  Thanks.
@Corsair-63  Camelo  In [Connect to GNSS](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:global_navigation_satellite_system#for_bluetooth_gnss_receivers) I've added information about connecting to GPS via bluetooth, does this look right?  If not please edit.
Also since you have RPI can you provide a screenshot for [Connecting to GNSS to Raspbery Pi](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:connect_to:global_navigation_satellite_system#for_bluetooth_gnss_receivers#to_raspberry_pi)? 
That will finish  Connect to GNSS!
Thanks.