OpenCPN/OpenCPN #3876: Missing translated strings

Corsair-63, 2024-05-01:
**Describe the bug**
There are some strings that are not translated (Spanish) but searched in Crowdin or they translated either slightly different in English also.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings' - 'Connections Tab'
2. 'Add Connection ...' / 'Edit connection ...' / 'Use magnetic bearing in output sentence APB'
2. deleting a track in confirm windows shows button 'Yes'
3. the strings in English did not exist as they are in the English version, e.g. connection haven't '...' and APB sentence the original English in Crowdin is 'ECAPB'
4. Related strings only displayed in English

**Expected behavior**
if the original matches with Crowdin file supposed to be displayed in the translation active in OCPN.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11 Pro 22631.3527 (x64 & ARM64)
 - Version: 5.9.0-0+22836ab Build 2024-04-28

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
Already done.

bdbcat, 2024-05-02:
These translatable strings will be added to the Crowdin translation workflow soon.

bdbcat, 2024-06-08:
Crowdin templates updated.

Corsair-63, 2024-06-08:
I've done the Spanish translation, please make a double check if would be any miss translation or not clear.

still have not been exchanged some string that have not proper translation as per Spanish official version of COLREG, and they are marked in translation as approved ones:
_Not Under Command_ - Sin Control ?? - don't know what does mean - Off. translation COLREG: **Sin Gobierno**
_Restricted Manoeuvrability_ - Maniobrabilidad Restringida, Not properly, - better: **Maniobra Restringida**
_Constrained by draught_ - Limitación por calado - Off. translation COLREG: **Restringido por su calado**
_Moored_ - Amarrado, not properly - Off. translation COLREG: **Atracado**

translations approved with meaning that could lead to error:
_Input message filtered, output message filtered and dropped_ 
translated as
Mensaje de entrada filtrado, mensaje de salida filtrado y dejado
translation most of the times very doubtful, better translation:
**Mensaje de entrada filtrado, mensaje de salida filtrado y rechazado.**

another one with same characteristics:
_Input Message filtered and dropped_
translated as
Mensaje de entrada filtrado y dejado
translation most of the times very doubtful, better translation:
**Mensaje de entrada filtrado y rechazado**

in both translations "dejado" in Spanish could be taken as allowed.

I've tried to exchange it but not allowed to do it either no answer because approved translations.

bdbcat, 2024-06-09:
Indicated ES translations updated as recommended.

Corsair-63, 2024-06-09:
It should be considered both in the original in English and in local translations to replace the term GPS with the more appropriate and current GNSS (Global Navigation Satellite System), as you well know GPS is a satellite positioning system of a country (USA) when most of our receivers are already multi-system.

leamas, 2024-06-13:
@Corsair-63 I assume that the basic issue reported here is fixed in the beta and that we thus can close this bug. 

Right?

Corsair-63, 2024-06-13:
Yes, the basic is done.

closed 2024-06-13 by Corsair-63