OpenCPN/OpenCPN #3878: Toolbar - Options -Connections sets size of Edit Connection Menu

rgleason, 2024-05-02:
**Describe the bug**
The size of the Connections Edit/Add Menu is set by Options - Connections. 
Shouldn't it have its own size settings, be resizeable and be remembered?

**To Reproduce**
[image]
[image]

**Expected behavior**
The edit menu to have its own size and location parameters, be resizeable and be remembered.

**Screenshots**
added

**Desktop (please complete the following information if applicable):**
 Win 11  5.9

bdbcat, 2024-05-03:
The connections edit dialog is modal, meaning that it takes over all OCPN UI when active.  So there is no reason to make it resizable.
I see that you are not using the very latest github master build.  I have reduced the dialog size somewhat.

NOTABUG, though might be called an RFE at a stretch.

rgleason, 2024-05-03:
Dave, I am looking at this as a user.  "connections edit dialog is modal"  means nothing to the user.

bdbcat, 2024-05-03:
I agree.  The user will find that the edit connection dialog must be closed (OK or Cancel) before anything else can be done with OCPN.  This is normal Windows behavior.
The question of size is purely cosmetic.  NOTABUG.

rgleason, 2024-05-03:
Ok, I thought I changed it to RFE. Will update O as I need it anyway, and see how it looks. Of course the edit menu needs  different sizes and there is more or less switch too, so it is not a simple interface to program.

closed 2024-05-07 by rgleason